tango-controls/cppTango #16: Cmake: disable -rpath option when compiling with cmake

bourtemb, 2016-09-21:
Hi,

In the current state, the compilation with Cmake is compiling with the -rpath option.
This option should be disabled as it was before when compiling with the original Makefile.
Thanks in advance.

Reynald

Ingvord, 2016-09-21:
As far I as remember rpath is being removed by cmake when the executable/library is being installed by default:

```
$ readelf -d install/lib/libtango.so.9.2.5 |head -20                  

Dynamic section at offset 0xaef340 contains 33 entries:
  Tag        Type                         Name/Value
 0x0000000000000001 (NEEDED)             Shared library: [libzmq.so.3]
 0x0000000000000001 (NEEDED)             Shared library: [libomniORB4.so.1]
 0x0000000000000001 (NEEDED)             Shared library: [libomnithread.so.3]
 0x0000000000000001 (NEEDED)             Shared library: [libCOS4.so.1]
 0x0000000000000001 (NEEDED)             Shared library: [libomniDynamic4.so.1]
 0x0000000000000001 (NEEDED)             Shared library: [libstdc++.so.6]
 0x0000000000000001 (NEEDED)             Shared library: [libm.so.6]
 0x0000000000000001 (NEEDED)             Shared library: [libgcc_s.so.1]
 0x0000000000000001 (NEEDED)             Shared library: [libc.so.6]
 0x000000000000000e (SONAME)             Library soname: [libtango.so.9]
 0x000000000000000c (INIT)               0x439940
 0x000000000000000d (FINI)               0x8dde7c
 0x0000000000000019 (INIT_ARRAY)         0xc90000
``` 

Currently rpath is used in the test executables:

```
$ readelf -d cpp_test_suite/asyn/asyn_cmd |head -20

Dynamic section at offset 0x8960 contains 35 entries:
  Tag        Type                         Name/Value
 0x0000000000000001 (NEEDED)             Shared library: [libdl.so.2]
 0x0000000000000001 (NEEDED)             Shared library: [libtango.so.9]
 0x0000000000000001 (NEEDED)             Shared library: [libzmq.so.3]
 0x0000000000000001 (NEEDED)             Shared library: [libomniORB4.so.1]
 0x0000000000000001 (NEEDED)             Shared library: [libomnithread.so.3]
 0x0000000000000001 (NEEDED)             Shared library: [libCOS4.so.1]
 0x0000000000000001 (NEEDED)             Shared library: [libomniDynamic4.so.1]
 0x0000000000000001 (NEEDED)             Shared library: [libstdc++.so.6]
 0x0000000000000001 (NEEDED)             Shared library: [libm.so.6]
 0x0000000000000001 (NEEDED)             Shared library: [libgcc_s.so.1]
 0x0000000000000001 (NEEDED)             Shared library: [libc.so.6]
 0x000000000000000f (RPATH)              Library rpath: [/storage/Projects/org.tango/git/cppTango/build]
```

In order to use *just built* library and not the one you might have in the library path (it will confuse) or even worth: not to use anything (if you have just cloned the repo and run the build for the first time)

__NOTE__ test executables ain't being installed and therefore *must* use libraries from the current build tree

It is possible to fully disable rpath, for instance see [this](https://cmake.org/Wiki/CMake_RPATH_handling)

But this will make development process a nightmare - imagine you have to configure library dependencies manually instead of letting cmake handle it for you

Correct me if I am wrong but I find this issue invalid

bourtemb, 2016-09-27:
You're right.
I didn't know cmake was removing rpath during the install phase.
I close this issue.

closed 2016-09-27 by bourtemb