tango-controls/cppTango #18: cmake: Support cmake v2.8.9

bourtemb, 2016-09-21:
Hi,

We have an issue compiling with CMake on our Debian7 machines where CMake 2.8.9 is installed.
We get the following error:
```
CMake Error at cpp_test_suite/CMakeLists.txt:10 (cmake_host_system_information):
  Unknown CMake command "cmake_host_system_information".
```

Ingvord, 2016-09-21:
Since supporting CMake 2.8.9 requires more work. I would suggest to postpone it till we fully migrate from svn (aka close #1 and #11)

bourtemb, 2017-12-05:
It is actually easy to install a more recent version of CMake manually (even the latest stable) on a Debian 7 computer as demonstrated in #417 so this requirement to support CMake v2.8.9 is not considered as critical.
Closing this issue. Please reopen if it's a real problem for you.

closed 2017-12-05 by bourtemb