tango-controls/cppTango #273: Can not compile libtango on Ubuntu 16 using libzmq3-dev package from apt

Ingvord, 2016-09-28:
Error message

```
[ 17%] Building CXX object cppapi/client/CMakeFiles/client_objects.dir/zmqeventconsumer.cpp.o
/home/ingvord/Projects/org.tango/git/cppTango/cppapi/client/zmqeventconsumer.cpp: In member function ‘virtual void* Tango::ZmqEventConsumer::run_undetached(void*)’:
/home/ingvord/Projects/org.tango/git/cppTango/cppapi/client/zmqeventconsumer.cpp:186:18: error: cannot convert ‘zmq::socket_t’ to ‘void*’ in assignment
  items[0].socket = *control_sock;
                  ^
/home/ingvord/Projects/org.tango/git/cppTango/cppapi/client/zmqeventconsumer.cpp:187:18: error: cannot convert ‘zmq::socket_t’ to ‘void*’ in assignment
  items[1].socket = *heartbeat_sub_sock;
                  ^
/home/ingvord/Projects/org.tango/git/cppTango/cppapi/client/zmqeventconsumer.cpp:188:18: error: cannot convert ‘zmq::socket_t’ to ‘void*’ in assignment
  items[2].socket = *event_sub_sock;
                  ^
/home/ingvord/Projects/org.tango/git/cppTango/cppapi/client/zmqeventconsumer.cpp: In member function ‘bool Tango::ZmqEventConsumer::process_ctrl(zmq::message_t&, zmq::pollitem_t*, int&)’:
/home/ingvord/Projects/org.tango/git/cppTango/cppapi/client/zmqeventconsumer.cpp:1063:47: error: cannot convert ‘zmq::socket_t’ to ‘void*’ in assignment
                 poll_list[old_poll_nb].socket = *tmp_sock;

```

Linux ingvord-VirtualBox 4.4.0-38-generic #57-Ubuntu SMP Tue Sep 6 15:42:33 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux

libzmq3-dev 4.1.4-7

Ingvord, 2016-09-28:
Current work-around:

Download, compile and install [zmq-4.0.5](https://github.com/zeromq/zeromq4-x/releases/tag/v4.0.5)

Download and copy into zmq-4.0.5 installation folder, e.g. /usr/local/zmq-4.0.5/include, *.hpp files from [cppzmq](https://github.com/zeromq/cppzmq)

jkotan, 2017-08-03:
I have the same problem for debian 9 (stretch).

Ingvord, 2017-08-03:
@jkotan did the workaround work out for you?

jkotan, 2017-08-03:
Sorry,  I've tried. I have switched with my tests from `stretch` back to `jessie` and decided to wait with them for a proper implementation for zmq in tango, i.e. an adapter or factory for different zmq APIs.

Ingvord, 2017-08-03:
I mean good point on adapter pattern

t-b, 2017-11-01:
Is that the offical zeromq API you are using? The socket type from https://github.com/zeromq/libzmq is void*.

andygotz, 2017-11-02:
For now only zmq 4.0.5 is officially supported.

Ingvord, 2017-11-02:
Well, as far as I remember. We are actually using https://github.com/zeromq/cppzmq which is not pure libzmq I believe, but a binding

andygotz, 2017-11-08:
@t-b  the situation is even more complicated. I just checked on  Ubuntu 17.10 and here tango is using libzmq.so.5.1.1 and works. So apparently my statement that tango has been tested only with 4.0.0 of zmq is not true. At least it works with more recent versions like 5.1.1. This does not mean it works with all recent versions. Let us know if it works with zmq 4.0.8.

bourtemb, 2017-12-08:
#421 addresses this issue.
The solution to this issue is to install [cppzmq](https://github.com/zeromq/cppzmq) (v4.2.2 for instance) and to use -DCPPZMQ_BASE=<cppzmq install base folder> when compiling libtango.
With this solution, Travis CI tango-9-lts tests are passing for libzmq3-dev Debian packages from wheezy-backports, jessie and stretch (Debian 7, Debian 8 and Debian 9).

PR #561 (2019-08-17): Be compatible with all in-use cppzmq versions.
Follow up from #530.

Close #499.
Close #273.
Close #535.
closed 2019-08-17 by t-b