tango-controls/cppTango #292: Please provide a patch for mysql 5.7 and mariadb-10 support

picca, 2016-10-23:
Hello, In order to upload tango9 into Debian,it would be great to have an official patch that I should apply to tango 9.2.2.

Debian 9 will provide only mariadb for security support reasons.
mysql 5.7 will be available only in Unstable.

the dead line for transition is realy realy close,  (around 5 november if I remember correctly).

PS; For now we have a broken tango8 which does not install anymore due to .sql script which does not support mysql 5.7.

thanks

Fred.

Ps; Which version of tango will you support during all the stretch lifetime (around 5 years) tango8 or tango9 ?

hardion, 2016-12-03:
I have made a patch on sourceforge, Issue 815. So this issue should be close by now.
Frederic, did you get it?
Tango maintainers, did you apply the patch?

/Vincent

Ingvord, 2016-12-05:
For the next 5 years we would like to support Tango 9.

bourtemb, 2016-12-08:
Hi Vincent,

It seems another approach was taken by giving a default timestamp value for the accessed timestamp instead of using your suggestion of swapping the fields.
You can have a look at the changes here:
[https://github.com/tango-controls/TangoDatabase/commit/c5915b3a2c5133e6f4e3416a8d2f293fc64b97f2](https://github.com/tango-controls/TangoDatabase/commit/c5915b3a2c5133e6f4e3416a8d2f293fc64b97f2)

hardion, 2016-12-08:
> tango-controls/TangoDatabase@c5915b3 <https://github.com/tango-controls/TangoDatabase/commit/c5915b3a2c5133e6f4e3416a8d2f293fc64b97f2>

Thanks Reynald.
The essential is to fix the problem, isn’t it?

Cheers,
/Vincent

closed 2017-01-20 by bourtemb