tango-controls/cppTango #314: ZmqEventSubscriptionChange segmentation fault

AntoineDupre, 2016-11-24:
We recently noticed a segmentation fault that affects all the tango servers. It happens when a client calls the dserver command ZmqEventSubscriptionChange with an invalid event type as a command argument.

Please find attached the gdb backtrace of the segfault generated after the following dserver command:
[tango_server_gdb_backtrace.txt](https://github.com/tango-controls/cppTango/files/611766/tango_server_gdb_backtrace.txt)

```
dserver/databaseds/2/ZmqEventSubscriptionChange

Argin: ["dserver/databaseds/2", "State", "subscribe", "typo"]
```

That might happen if a client subscribes to an attribute with the stateless option, while the endpoint given by the ZmqEventSubscriptionChange command is not accessible (in our case due to a network configuration).

For some reasons, several subscription commands will run successively, each with a different event type argument:

- change
- idl5_change
- idl5_idl5_change

The latest does not exist and generates the segmentation fault.

Tested with tango 9.2.2.

bourtemb, 2016-11-24:
Hi Antoine,

In theory, you don't need to (and you probably shouldn't) call this method directly...
Why are you calling it directly?
The Tango kernel will invoke this method and will always do it with valid arguments (at least it should. if it's not the case, this is indeed a bug).
We could indeed improve the error handling but the fact that you are calling this method directly makes me think you are doing something you should probably not do.

bourtemb, 2016-11-24:
Hi,

I think I probably misunderstood your use case. You are probably not calling this method directly. This is simply where it crashes.
Could you please give us an example of a simple client reproducing the problem or at least give us the subscribe_event methods lines and the arguments you are using in your client so we could try to reproduce the problem?

Kind regards,
Reynald

AntoineDupre, 2016-11-25:
Hi,

As you said, I am calling this dserver command to provide a minimalist use case to reproduce the segfault.

I guess we can see this issue as two different problems:

1. A typo in the ZmqEventSubscriptionChange command makes the tango device server segfault.
2. A client can call ZmqEventSubscriptionChange command (through the subscription API) with an  unknown event type name. 

------------------------------------------------------------------------------

The first point can be reproduced with a basic PyTango client:

```python
from tango import DeviceProxy

ds = DeviceProxy("sys/tg_test/1")
# enable polling for event subscription
ds.poll_attribute("State", 3000)
dserver = DeviceProxy("dserver/TangoTest/test")
args = ["sys/tg_test/1", "state", "subscribe", "typo"]
dserver.command_inout("ZmqEventSubscriptionChange", args)

```
------------------------------------------------------------------------------

The second one is more difficult to reproduced because it requires a client, ables to communicate with a tango device server, but the ZMQ endpoints have to be unreachable for it (In my case, this has been triggered by a special network configuration).

With such kind of network configuration, a PyTango client should be able to trigger the server segfault after few seconds:
```python

from tango import DeviceProxy, EventType

ds = DeviceProxy("sys/tg_test/1")
ds.poll_attribute("State", 3000)

def callback(e):
    print(e)

ds.subscribe_event("State", EventType.CHANGE_EVENT, callback, stateless=True)
```

By inspecting the client subscription request; it appears that the server segfautl is triggered after the third connection request:

1. ZmqEventSubscriptionChange(["sys/tg_test/1", "state", "subscribe", "change"])
2. ZmqEventSubscriptionChange(["sys/tg_test/1", "state", "subscribe", "idl5_change"])
3. ZmqEventSubscriptionChange(["sys/tg_test/1", "state", "subscribe", "idl5_idl5_change"])

We guess that the client subscription mechanism is something like:

1. the subscription command returns some endpoints but the client can not reach them and it failed to access the ZMQ channel.

2. As we run with the stateless option, and as we previously failed to access the channel (but the subscription command worked), the client tries again to subscribe to the event channel but with the IDL type. It adds "idl5_" before the previous event type name. As previously, the subscription command returns some endpoints that are not accessible.

3. As previously, we are in a case were the last subscription command works but the client failed to access the ZMQ channel. The client add (again) "idl5_" at the beginning of the event type name and sends again the subscription command. 
Now the event type is unknown ("idl5_idl5_change"), the server segfault.

I guess, this behavior fit with [client/event.cpp (line 1617)](https://github.com/tango-controls/cppTango/blob/master/src/client/event.cpp)  when `event_name` already starts with "idl5_": 

```cpp
int EventConsumer::subscribe_event(

....
//
// Change event name if it is IDL 5 compatible:
// This code is Tango 9 or more. If the remote device is IDL 5 (or more), insert tango IDL release number
// at the beginning of event name.
//
...
        if (dd_extract_ok == true && add_compat_info == true && dvlsa->lvalue[1] >= MIN_IDL_CONF5)
        {
                event_name = EVENT_COMPAT_IDL5 + event_name;
...
```

Cheers,
/Antoine

commit cea3f7bd (2016-12-15): Fix#314 master (#336)

* Fix #314: Check ZmqEventSubscriptionChange command event parameter
commit 790fc47f (2016-12-16): Fix #314: Check ZmqEventSubscriptionChange command event parameter
closed 2016-12-15 by Ingvord