tango-controls/cppTango #315: Events do not work when subscribing to events with FQDN containing localhost

bourtemb, 2016-11-25:
This issue is described in the following Tango forum topic:
http://www.tango-controls.org/community/forum/c/development/python/event-system-problem

The events do not work as expected when a client subscribe to events using FQDN starting with localhost.
A workaround is to use the real machine name instead of localhost.

This was reproduced in C++.
To be checked in Java too.

andygotz, 2017-10-25:
A first attempt at solving this problem was done in the PR #385 but @Ingvord proposes to not use this PR and to start a new one.

Ingvord, 2017-10-30:
@bourtemb, could you please test #404 in the conditions mentioned in this issue? If it is OK we can merge #404 into the master and backport it into tango-9-lts

bourtemb, 2018-05-15:
This should be fixed in #423 (available in cppTango 9.3.0).
Closing this issue. Please reopen if you still experience this problem with cppTango >= 9.3.0.

closed 2018-05-15 by bourtemb