tango-controls/cppTango #316: Deadlock between event subscription and event callback

vxgmichel, 2016-11-25:
*This issue has already been reported on sourceforge as [ticket #830](https://sourceforge.net/p/tango-cs/bugs/830/)*.

I noticed that event subscriptions and event callbacks share a hidden lock, which can cause a deadlock in the following situation:

```python
import time
import tango
import threading

CHANGE_EVENT = tango.EventType.CHANGE_EVENT

def callback(event):
    with monitor:
        print(event.attr_value.value)

monitor = threading.RLock()
with monitor:
    proxy = tango.DeviceProxy('sys/tg_test/1')
    eid1 = proxy.subscribe_event('double_scalar', CHANGE_EVENT, callback)
    time.sleep(1.)
    eid2 = proxy.subscribe_event('State', CHANGE_EVENT, callback)
    print('OK')
raw_input()
```
If an actual `double_scalar` event is generated during `time.sleep`, here's what happens:
- the callback thread acquires the hidden lock
- the callback thread waits for the monitor lock
- the main thread wakes up (end of sleep)
- the main thread runs subscribe_event
- the main thread waits for the hidden lock
- deadlock!

This is actually what happens in some devices, when the event callback tries to acquire the monitor lock (useful to prevent race condition between device callbacks and event callbacks).

Weirdly enough, I haven't been able to reproduce it during the first `init_device`. Instead, I had to run the `INIT` command to run `init_device` a second time and reliably generate the deadlock.

Tested with tango 9.2.2.

bourtemb, 2018-01-31:
Indeed there is a ReaderWriter lock shared between the event callbacks and the event subscriptions.
See [event.cpp:1255](https://github.com/tango-controls/cppTango/blob/tango-9-lts/cppapi/client/event.cpp#L1255) and [zmqeventconsumer.cpp:1953](https://github.com/tango-controls/cppTango/blob/tango-9-lts/cppapi/client/zmqeventconsumer.cpp#L1953).

bourtemb, 2018-01-31:
In your example, the problem occurs because the callback could be blocked waiting for the lock.
In your example, it is blocked because it is trying to acquire a lock taken by another thread (the thread which is doing the subscribe_event calls).
Ideally, a callback should not take too long to execute and should not block.
The solution proposed by Tiago in [this forum post](http://www.tango-controls.org/community/forum/c/development/python/event-subscription-in-a-pytango-device/?page=1#post-728) is actually a good solution:

> My workaround for the problem is to have a worker thread waiting for jobs.
Anytime I have a tango blocking call I throw it into the worker thread.

bourtemb, 2018-03-19:
@vxgmichel, did my last comments answer your question?
Do you consider the work-around suggested by Tiago on the tango-controls forum a good solution to solve this issue?
Can we close this issue?

vxgmichel, 2018-03-22:
@bourtemb Sorry for the late answer.

It is indeed possible to avoid this issue by not acquiring any lock in the callback. And since it is hard to make sure that this is the case, the thread pool solution is probably the safest choice. However, that means the application now has the responsibility to start and tear down the pool. In my opinion, this extra burden on the application could be avoided by releasing the ReaderWriter lock before running the callback. 

Now, I understand that this mechanism might be tricky to change and I won't mind if you close and tag the issue as *won't solve*.

closed 2019-07-17 by andygotz