tango-controls/cppTango #324: CMake compilation failure of tango-9-lts

grm84, 2016-12-06:
$ cmake --version
cmake version 2.8.11
$ cmake .. -DOMNI_BASE=~/packages/omniORB-4.2.1 -DZMQ_BASE=~packages/zeromq-4.0.7 -DCMAKE_BUILD_TYPE=RELEASE -DCMAKE_VERBOSE_MAKEFILE=true -DIDL_BASE=~/software/tango-idl
-- The C compiler identification is GNU 4.8.5
-- The CXX compiler identification is GNU 4.8.5
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
System TANGO_HOST=tcfidell11:10000
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.27.1")
-- Found Doxygen: /usr/bin/doxygen (found version "1.8.5")
Using OMNI_BASE=~/packages/omniORB-4.2.1
Using ZMQ_BASE=~packages/zeromq-4.0.7
Verifying ~/packages/omniORB-4.2.1/bin/omniidl
found GNU compiler ...
setup for release build ...
CMake Error at configure/CMakeLists.txt:70 (add_compile_options):
  Unknown CMake command "add_compile_options".
Call Stack (most recent call first):
  CMakeLists.txt:18 (include)

bourtemb, 2016-12-06:
Indeed CMake 2.8.11 does not support _add_compile_options_ CMake command.
tango-9-lts branch should compile using the autotools as before on a machine having a too old version of CMake installed.
For the moment, CMakeLists.txt defines:
cmake_minimum_required(VERSION 2.8) 
This is not correct since it Cmake 2.8.11 is not able to compile the code. The minimum required version should be updated to something which makes sense (To be defined and tested).
What we know for sure is that we can use CMake 3.0.2 to compile successfully.

Ingvord, 2016-12-06:
@grm84 Please use code highlighting when posting console output:

```
Not formated
```

i.e. wrap with ```

I mark this as duplicate to #18

Ingvord, 2016-12-06:
Maybe it is better to leave duplicates open till we fix the original one and then close all related issues.

t-b, 2017-10-30:
See https://stackoverflow.com/a/23995391, basically one just has to use `CMAKE_CXX_RELEASE/CMAKE_CXX_DEBUG` etc. flags manually instead of `add_compile_options`.

PR #422 (2017-12-08): Set CMake minimum requirement to CMake 2.8.12
Fix #324
commit c99c3ea6 (2017-12-08): Set CMake minimum requirement to CMake 2.8.12 (#422)

Fix #324
closed 2017-12-08 by bourtemb