tango-controls/cppTango #339: pointer-cast-size-mismatch attrgetsetprop.cpp:200

picca, 2016-12-17:
Hello,

while preparing the tango package of the 9.2.5~rc5, I got this error.

You can find other warning errors here, which are more packaging related.

https://qa.debian.org/bls/packages/t/tango.html

Cheers

Ingvord, 2016-12-17:
Is this urgent? I mean, is it possible to provide debian package with this error?

picca, 2016-12-17:
> Is this urgent? I mean, is it possible to provide debian package with this error?

I think that plenty of packages are distributed with bug :), so it is possibleto provide tango 9.2.5 with this bug.

Now I do not know if this can be fixed with a one liner for the final release of 9.2.5 in january.

Cheers

Fred

andygotz, 2017-10-25:
Needs to be done in LTS9 as well. If done then please close this issue. Thanks!

bourtemb, 2017-10-26:
Done on Tango 9 LTS too (PR #406).
Closing this issue.

closed 2017-10-26 by bourtemb