tango-controls/cppTango #349: server_set_event_loop() not being run in Windows

mikesmith-at-nrc, 2017-01-19:
I have a tango server where the event loop runs just fine when compiled for Linux, but does not run at all in Windows.

When debugging the issue, it looks like the code in Util::server_run() surrounding the orb->run() call is very different depending on if _TG_WINDOWS_ is defined:

```
#ifdef _TG_WINDOWS_
   <snip>
   try
   {
      orb-run();
      server_cleanup();
   <snip>
#else 
   <snip>
   try
   {
      if (ev_loop_func != NULL)
      {
         <while loop that runs event loop and calls orb->perform_work()>
      }
      else
      {
         orb->run();
      }
      server_cleanup();
   <snip>
#endif
```

Is there a reason for not using the event loop functionality in windows? Would a cut/paste of the if(ev_loop_func) block fix this issue?

bourtemb, 2017-01-20:
Thank you very much for creating this ticket.
It looks like the implementation of this user defined event loop has been forgotten on Windows indeed.
The cut & past you've suggested might work. Some tests are needed.

mikesmith-at-nrc, 2017-02-09:
I have a patch that fixes this issue, how should I submit it?

bourtemb, 2017-02-10:
Hi,
Great!
If you could create a Pull Request following the instructions from 
[CONTRIBUTE](https://github.com/tango-controls/cppTango/blob/master/CONTRIBUTE.md) file, that would be the best.
If you need more details on the process, this page might help you:
https://gist.github.com/Chaser324/ce0505fbed06b947d962

If your patch is for tango 9 LTS, replace master with tango-9-lts in the instructions from the above links.
You can also attach the patch to this ticket if you prefer but we encourage contributions via Pull Requests because it keeps track of the contributors and simplify the review process.
The best would be to get a Pull Request for the master branch (future Tango V10) and for TANGO 9 Long Term Support branch (tango-9-lts) but if you could already submit PR for one of the 2, that would be great!
Thanks for your contribution!
Reynald

Ingvord, 2017-02-16:
Done in #352

commit a9ccc997 (2017-02-16): resolve #349: add missing event loop implementation within _TG_WINDOWS_ ifdef
closed 2017-02-16 by Ingvord