tango-controls/cppTango #350: Case sensitivity error when subscribing to a change event

gcerretani, 2017-01-30:
When I subscribe to a change event, in the first event received at the subscription time the attr_value.name value has the right case sensitivity, as it was in PyTango 8. Then, the following change events have that field lower cased.

In the PERIODIC_EVENT, that field is always lower cased. It would be nice to get the not lowered name, as in PyTango 8.

Example:

- first event:
```
EventData[
     attr_name = 'tango://...:20000/satserver/lab/1/g2'
    attr_value = DeviceAttribute(data_format = tango._tango.AttrDataFormat.SCALAR, dim_x = 1, dim_y = 0, has_failed = False, is_empty = False, name = 'G2', nb_read = 1, nb_written = 1, quality = tango._tango.AttrQuality.ATTR_VALID, r_dimension = AttributeDimension(dim_x = 1, dim_y = 0), time = TimeVal(tv_nsec = 0, tv_sec = 1484308260, tv_usec = 881227), type = tango._tango.CmdArgType.DevDouble, value = 8.0, w_dim_x = 1, w_dim_y = 0, w_dimension = AttributeDimension(dim_x = 1, dim_y = 0), w_value = 0.0)
        device = SatServer(satserver/lab/1)
           err = False
        errors = ()
         event = 'change'
reception_date = TimeVal(tv_nsec = 0, tv_sec = 1484308261, tv_usec = 822368)]
```

- standard event:

```
EventData[
     attr_name = 'tango://...:20000/satserver/lab/1/g2'
    attr_value = DeviceAttribute(data_format = tango._tango.AttrDataFormat.SCALAR, dim_x = 1, dim_y = 0, has_failed = False, is_empty = False, name = 'g2', nb_read = 1, nb_written = 1, quality = tango._tango.AttrQuality.ATTR_VALID, r_dimension = AttributeDimension(dim_x = 1, dim_y = 0), time = TimeVal(tv_nsec = 0, tv_sec = 1484308266, tv_usec = 881001), type = tango._tango.CmdArgType.DevDouble, value = 7.96219999999992, w_dim_x = 1, w_dim_y = 0, w_dimension = AttributeDimension(dim_x = 1, dim_y = 0), w_value = 7.0)
        device = SatServer(satserver/lab/1)
           err = False
        errors = ()
         event = 'change'
reception_date = TimeVal(tv_nsec = 0, tv_sec = 1484308266, tv_usec = 929997)]
```

The problem seems related to the C++ side of the client. More information are found in the bug [#87](https://github.com/tango-controls/pytango/issues/87) closed in the PyTango repository.

bourtemb, 2017-02-02:
Thank you very much for the bug report.
It seems this bug was introduced due to some changes related to the forwarded attributes in cppapi/client/zmqeventconsumer.cpp file.
(https://github.com/tango-controls/cppTango/commit/c5d7ba8801b0f9fdb38cf2c1034b5e70783ac6f6#diff-b0556cfbe11b1b16f9c0b4e3ac536871 - lines 2285 and 2286 of this commit) 
Commenting lines 2432 and 2433 of cppapi/client/zmqeventconsumer.cpp file in tango-9-lts latest version seems to solve the issue for simple cases but might trigger issues with forwarded attributes as the comments above these lines are suggesting.
Deeper investigations will be needed.

bourtemb, 2019-03-29:
This should be fixed in cppTango v9.3.3

HFRibeiro, 2020-01-13:
Hi @bourtemb ,
I'm using
```
PyTango 9.3.1 (9, 3, 1)
PyTango compiled with:
    Python : 3.7.3
    Numpy  : 1.17.2
    Tango  : 9.3.3
    Boost  : 1.67.0
```

And still when subscribing to CHANGE_EVENT the first time the attribute comes with case sensitive something like `RandomAttr` but after that on events just `randomattr`.

Any idea what I might be doing wrong?

bourtemb, 2020-01-13:
> Any idea what I might be doing wrong?

Nothing!
I think I got confused at some point because I was looking at the `attr_name `which seems to be always in lower case. (I was double confused it seems because the issue was requesting to have the attribute name **not** in lower case...).
But the issue is actually about the `name ` field of the `attr_value`. 
The "_problem_" seems to be still present indeed.
Sorry for the confusion.
The concept of Attribute Name is in any _case_ (if I can say ;-) ) , _case_ insensitive so the current behaviour is not completely wrong. I can understand it can be annoying in some use cases though and maybe PyTango code is relying more on this case sensitivity?

bourtemb, 2020-01-13:
The problem is actually not present when subscribing to events via an AttributeProxy instead of via a DeviceProxy.

bourtemb, 2020-01-13:
> The problem is actually not present when subscribing to events via an AttributeProxy instead of via a DeviceProxy.

Well, I would say the problem is different when using an AttributeProxy. In this case, the first event is with the case returned when doing a typical read_attribute (first event received by the callback is the result of a synchronous read_attribute and does not come from ZMQ).
The following events are then in the case specified by the user in the subscription command.

HFRibeiro, 2020-01-14:
Ok, thank you very much for the explanation.
Indeed in https://tango-controls.readthedocs.io/en/latest/development/general-guidelines/naming.html 
it says 

> The device name, the command name, the attribute name, the property name, the device alias name and the device server name are case insensitive.

So you are right this shouldn't be a problem, the name should come as case insensitive.
The clarification was very important for me at least! 
Thank you