tango-controls/cppTango #351: Support DevEnum in Commands

josefcibulka, 2017-02-10:
Support for DevEnum for Command parameters is incomplete, making it unusable. It is possible to choose DevEnum as the type of input and output in Pogo, but there is no way to set the list of enum labels. This is not only a problem of Pogo, but I do not even see any equivalent of Tango::Attribute::enum_labels in Tango::Command.

PS: I would also like to see #257 implemented and since it also requires changes in the Tango::Command class, I think it would make sense to do them at the same time.

Ingvord, 2017-06-04:
@josefcibulka we have merged #357 into the master branch  which resolves this issue. Could you please test it either building from source or using our debian repo: [link](https://bintray.com/tango-controls/debian/cppTango/10.0.1)? If you confirm that it is OK we will close the issue. Thanks in advance.

josefcibulka, 2017-07-11:
@Ingvord  Apologies for the long delay. I confirm that it works well for me. Am I correct that the latest versions of Pogo and ATK panel do not support it yet? Should I open a ticket in ATK?

Ingvord, 2017-07-12:
@josefcibulka thanks for the positive response. Yes, please open issues in the corresponding projects.

closed 2017-07-12 by Ingvord