tango-controls/cppTango #361: missing attr_conf event after "Reset to default value"

gscalamera, 2017-04-12:
Using Jive, after right click on "Attribute config"->Description selecting "Reset to default value", the attr_conf event with the default value for the description is not sent

bourtemb, 2017-07-04:
Thanks Graziano for the report.
What jive does when the user clicks on "Reset to default value" is the following:

1. Delete the description for this attribute in the Database
2. Restart the device

From the tests I've done with a C++ event subscriber talking to a C++ server using Tango 9.2.5a, it looks like the event subscriber no longer receives any attribute configuration event after a restart device.

mliszcz, 2019-12-17:
Hi @bourtemb and @gscalamera 

I don't think this is a bug. Reynald already explained the behavior on Jive side.

The usual way of triggering Attr_Conf event from external client (like Jive) is to call CORBA method set_attribute_config_5. The server will then push Attr_Conf event to any registered clients (**regardless if there was a change in properties or not**).

What we can do:
1. At the very end of `DServer::restart` push events regardless if there was a change or not (consistent with current behavior of set_attribute_config_5)
2. or, before device is deleted, store current property values and after the restart fire events only for modified attributes.(probably more reasonable)

For option 1., it is as simple as:
```cpp
for (Attribute* attr : new_dev->get_device_attr()->get_attribute_list())
{
    new_dev->push_att_conf_event(attr);
}
```

Options 2. should not be much more complex.

What do you think?

lorenzopivetta, 2019-12-17:
+1 for option 1. Much cleaner, and I think it is reasonable clients get all-fresh events after a device restart... my 2c.

bourtemb, 2019-12-17:
+1 for option 1 too.

mliszcz, 2019-12-17:
Thanks Lorenzo and Reynald. I have PR and testcase ready. I'll wait with opening a PR until #573 is merged (it's a dependency required to pass the test and it will trigger merge conflicts).

PR #694 (2020-04-23): Send attribute configuration event to all clients after device restart
Fixes #361
closed 2020-04-23 by t-b