tango-controls/cppTango #365: migrate API doc to sphinx

Ingvord, 2017-05-02:
This issue is related to [issue 20 in tango-doc](https://github.com/tango-controls/tango-doc/issues/20)

Ingvord, 2017-07-20:
As the first step we can integrate with read-the-docs generating doxygen via sphinx (breath?)

andygotz, 2017-10-25:
Has something been done for this issue?

Ingvord, 2017-10-25:
Nope. I asked @piogor to help bridging doxygen and sphinx. Here is what he answered:

> It was decided to remove integration with sphinx from the current version of documentation [tango-doc]. It was mainly to some restriction in breath in how it generate the documentation. It generates either flat document with all classes, namespaces and so(really hard to browse) or documentation for specific class (which is hard to be useful) . Other issue was time it adds to processes at RtD and number of warnings generated.

So I guess we can not just wrap doxygen with sphinx. The whole documentation must be rewritten. Huge job :(

t-b, 2019-02-04:
Do we want take a stab at this again? I'm using doxygen/sphinx/breathe quite succesfully for other projects like http://docs.byte-physics.de/igor-unit-testing-framework/, which are of course smaller than tango.

t-b, 2019-07-30:
@piogor @bourtemb  @Ingvord 

I've took a first stab at this. And also tried something new. Instead of breathe (not really maintained anymore), I've played around with https://github.com/vovkos/doxyrest. This is a self-proclaimed successor of breathe and much more customizable.

The current version can be seen at [here](https://byte-physics.de/public-downloads/cppapi-doxyrest/index.html), I did not yet adapt any doxygen comments or did heavy tweaking. For comparison the doxygen html files are [here](https://tango-controls.github.io/cppTango-docs/).

Opinions?

Ingvord, 2019-07-30:
@t-b, 

thanks for sharing! Looks interesting. The big picture definitely makes sense to me. The Evil always hides in the details though, so we need to see how it will perform. 

The main goal is to integrate documentation generation into CI e.g. build it with every commit into master/or when tag is set... so if we can achieve that with this framework I will be more than happy!!!

t-b, 2019-07-30:
@Ingvord My plan is to have a docker image with the full documentation toolchain and let travis do the building and pushing.

bourtemb, 2019-07-31:
The first results look promising indeed. 
While going quickly through this generated documentation, I noticed that the link "Related pages" in the following page does not seem to work as expected. In the past it linked to a page like
https://tango-controls.github.io/cppTango-docs/pages.html
Now, it looks like this page has become a section of the [index.html](https://byte-physics.de/public-downloads/cppapi-doxyrest/index.html) page.

There is some work to be done on the following page:
https://byte-physics.de/public-downloads/cppapi-doxyrest/page_recon.html
which is the equivalent of the following page:
https://tango-controls.github.io/cppTango-docs/recon.html 

But again, the first results look very promising. The search feature seems to work fine too.

t-b, 2019-08-01:
@Ingvord @bourtemb Thanks for having a look.

Yes some work needs to be done. This is mostly about doxygen comments having embedded HTML, can be found with `git grep -P -i -e "</[a-z]+>" --and -e "(//|\*)" "*.h"`, I would definitly fix that as well.

closed 2022-02-24 by t-b