tango-controls/cppTango #368: Pushing events with a value for state triggers a memory leak

vxgmichel, 2017-05-19:
I noticed a two issues with the pushing of events with values for state and status:

- values, stamps and qualities are ignored
- a memory leak is triggered

It can be reproduced with the following code:

```python
#! /usr/bin/python
from tango import DevState, AttrQuality
from tango.server import Device, command

class TestState(Device):

    def init_device(self):
        self.set_state(DevState.OFF)
        self.set_change_event('State', True, False)

    @command(polling_period=100)
    def update(self):
        self.push_change_event(
            'State', DevState.OFF, 1.0, AttrQuality.ATTR_VALID)

if __name__ == '__main__':
    TestState.run_server()
```

grm84, 2017-09-18:
I have also seen this problem pushing state, which I have tracked down, with valgrind, to code in attrSetVal.cpp
`
==1555== 504 (432 direct, 72 indirect) bytes in 18 blocks are definitely lost in loss record 9,586 of 10,724
==1555==    at 0x4C29203: operator new(unsigned long) (vg_replace_malloc.c:334)
==1555==    by 0xF54D61E: Tango::Attribute::set_value(Tango::DevState*, long, long, bool) (attrsetval.cpp:1460)
==1555==    by 0xEAB3BCC: __set_value_scalar<19l> (attribute.cpp:86)
==1555==    by 0xEAB3BCC: PyAttribute::__set_value(std::string const&, Tango::Attribute&, boost::python::api::object&, long*, long*, double, Tango::AttrQuality*) (attribute.cpp:322)
==1555==    by 0xEAB4350: PyAttribute::set_value(Tango::Attribute&, boost::python::api::object&) (attribute.cpp:348)
==1555==    by 0xEB20AB7: PyDeviceImpl::push_change_event(Tango::DeviceImpl&, boost::python::str&, boost::python::api::object&) (device_impl.cpp:229)
`

bourtemb, 2018-02-06:
State and Status are special attributes because they existed even before the attributes were invented in Tango.
They were only commands before. Now they are commands and attributes.
They are also special because they are the only attributes which must be present in every Tango device class. So they are treated differently in the code.
By looking at [attribute.tpp:1371](https://github.com/tango-controls/cppTango/blob/tango-9-lts/cppapi/server/attribute.tpp#L1371) and [attribute.cpp:3162](https://github.com/tango-controls/cppTango/blob/tango-9-lts/cppapi/server/attribute.cpp#L3162), one can see the State and Status quality is forced to ATTR_VALID when an event is pushed (These methods are invoked by the fire_xxx_event() methods) so the way it is currently implemented, no matter what quality factor is passed to push_xxx_event() methods, the quality factor in the event will always be valid for State and Status attributes. This might be intentional since they are also commands, and commands do not have quality factors....

While investigating on this issue, I also noticed the DevState (resp. DevString) value passed to push_change_event() is actually ignored for State (resp. Status) attribute. The value taken into account is the current state (resp. status) of the device which you can get with get_state() (resp. get_status()) method.
The timestamp passed to push_xxx_event() method is also ignored with state and status and the current date and time seems to be used instead.

About the memory leak, this leak is due to the fact that when invoking push_change_event(), the DevState value given by the user is stored into a sequence when push_change_event() invokes Attribute::set_value(). This sequence is not deleted as it should in fire_change_event() where there is a nice if, which makes the code refusing to delete the sequence when the attribute name is State or Status. See [attribute.cpp:3839](https://github.com/tango-controls/cppTango/blob/tango-9-lts/cppapi/server/attribute.cpp#L3839) for instance.
This if in fire_change_event() is suspicious and seems to suggest that the sequence should never have been allocated for State and Status attributes, so this means that Attribute::set_value() should never be invoked by push_xxx_event() if the attribute name is state or status... It looks like this was the intended philosophy behind the current implementation (to be confirmed), even tough this does not seem to be documented.
I guess this was done like that to ensure that the State and Status events sent are matching what a client could see if he would invoke the State or Status command. The problem is that it is not documented, confusing and also I think it does not make sense to force the timestamps to the current date and time when pushing state and status events. And I think we should be able to choose the quality factor too.

Ingvord, 2018-12-04:
@bourtemb , thanks again for very detailed analysis of the issue!!!

closed 2022-02-24 by t-b