tango-controls/cppTango #372: Windows Compilation

NexeyaSGara, 2017-06-15:
Hi,

I'm getting a lot of errors while trying to build the Windows version with the new directory structure.

It's like I'm getting circular dependecies and /or namespaces issue.

I think that the visual compiler isn't as gentle as gcc on this thing.

It's happening for the "include_objects" parts, relates to the CMakesList in {TANGO}/include/

Do you have a tool for checking this ?

Thanks

Ingvord, 2017-06-16:
@NexeyaSGara I do not think that this is due to the new directory structure, because we have not touched internal dependencies. It is most probably due to CMake itself, as I believe before everybody used these dedicated VS projects maintained by @taurel (Correct me if I wrong). 

Unfortunately I do not know any good tool to check all of this. And since the codebase has really odd structure sometimes (like `#include` is used to literally include pieces of code of a file into another one which is, if rearranged, may cause compilation failures). The only way I see is to manually fix this.

Ingvord, 2017-06-16:
This one is related to #10

NexeyaSGara, 2017-06-16:
Hi,

The old windows solutions were including all the differents headers files directly.

It's Cmake that generates a lot of simple small projects depending on our CMakelists files.

It's in the include Cmakelists projects on my side that this problem occurs. I'm getting "undefined symbol" errors, like the compiler can't really find the right class to use.

But there is clearly something to "refactor" with the namespace Tango::, to be more clean. Will try yo narrow the research, but I can give you the last log if you want to have a view on my currents errors.

[log.txt](https://github.com/tango-controls/cppTango/files/1080014/log.txt)

Thanks

Ingvord, 2017-07-26:
@NexeyaSGara should we close this issue? I believe it is already outdated due to [this Appveyor output](https://ci.appveyor.com/project/NexeyaSGara/cpptango)

NexeyaSGara, 2017-07-31:
yes you can.

Ingvord, 2017-07-31:
Outdated

closed 2017-07-31 by Ingvord