tango-controls/cppTango #373: Data ready events not received for attribute names longer than 17 characters

t-b, 2017-06-19:
Steps to reproduce:
- Download DS and client from www.byte-physics.de/Downloads/tango-bug-server.tar.gz
- Compile MyDS
- Compile client with "g++ -I /usr/local/include/tango -ltango -lomniORB4 -lomnithread client.cpp -o a.out"
- Declare MyDS as `test/MyDS/1`
- Start MyDS
- Execute client `./a.out`

Client output:
```
Called for: a with counter = 0
Called for: a with counter = 1
Called for: a with counter = 2
```

Notice that only events for attribute `a` are received, events for attribute `aaaaaaaaaaaaaaaaaa` are never received.

I suspect that the length of `aaaaaaaaaaaaaaaaaa` is the problem.

Reproducible on debian stretch using tango 8.1.2. The tar file includes a log file produced via `make && ./bin/MyDS i1 -v100 2>1 > logfile.txt`.

bourtemb, 2017-06-21:
Hi Thomas,
I just tried with Tango 9.2.5a and the problem is not present. It looks like this was fixed at some point after tango 8.1.2.
You know you should be able to get tango9.2.5a from the official Debian packages on stretch?

closed 2017-06-21 by bourtemb

t-b, 2017-06-21:
@bourtemb I can't just switch to tango 9. Can you give me the exact commit which fixed the bug? Maybe I can backport the fix.

bourtemb, 2017-06-22:
I don't know the exact commit which fixed the bug and the guru who probably fixed this bug doesn't remember about this problem. So this might take some time to know where this bug is coming from...
Why can't you switch to tango9? You know Tango is backwards compatible so you can have several Tango versions running on your control system. A Tango 9 client can talk to a Tango 8 (and 7,...) server. A Tango 8 client can talk to a Tango 9 server.
It's just a matter of regenerating the code of the server sending data ready events with Pogo and run your server with Tango 9 libraries. You will need to upgrade your database server (and DB schema) to tango 9 too.
If this bug is on the client side, then you would need to recompile your clients with Tango 9 of course...
This is probably faster than finding the exact commit which fixed the bug and backporting the fix to Tango 8?

t-b, 2017-06-22:
> This is probably faster than finding the exact commit which fixed the bug and backporting the fix to Tango 8?

Industry does not work like that. At some point you decide to use version X and then you use that. Only if you really have to upgrade, ,you upgrade. Upgrading to tango9, without any compatibility bugs, is quickly a day.

bourtemb, 2017-06-22:
I understand. Tango 8 is not an LTS version so I would definitely recommend upgrading to Tango 9. We won't spend too much time trying to fix bugs which are already fixed in Tango 9. Sorry.
Tango is Open Source, you can have a look and try to find and fix the bug yourself in Tango 8 but we are already very busy in trying to fix other bugs. I'm sorry but I cannot help more on this subject.