tango-controls/cppTango #381: DServerSignal prevents CLion from setting breakpoint

Ingvord, 2017-07-20:
If DServerSignal is used CLion can not set breakpoint (Command timeout) while debugging

DServerSignal must be excluded conditionally from the runtime

bourtemb, 2018-11-15:
Is this issue related to SIGPIPE signal received while debugging a device server when a clients sends a command to this DS and a breakpoint is hit. If the client command triggers a timeout, a SIGPIPE signal occurs and it is impossible to continue to debug correctly?
If this issue was created for this specific problem, a workaround is to execute 
`handle SIGPIPE nostop noprint pass`
command in gdb.
This can be done automatically by putting this command into your ~/.gdbinit file (thanks @taurel  for the advice!).

bourtemb, 2019-07-17:
I created https://github.com/tango-controls/tango-doc/issues/287 to improve the doc with the 
`handle SIGPIPE nostop noprint pass`
tip.
@ingvord, was your original problem linked to that?
If yes, I think we can close this issue and follow the documentation update in https://github.com/tango-controls/tango-doc/issues/287

Ingvord, 2019-07-17:
@bourtemb , frankly speaking, I do not remember. The were two problems actually:

1) when client closes connection due to time out when server is in debug session - this can be work arounded as you described. So not an issue.

2) When one sets a breakpoint, DServerSignal considers it as an interruption and kills the server - can be work arounded by manually disabling signal handlers.

So in both cases we can close this issue and mark it as *won't fix*

closed 2019-07-17 by Ingvord