tango-controls/cppTango #386: Device ALARM state issue with forwarded attribute associated with a polled root attribute

bourtemb, 2017-07-28:
Let's consider the following setup:
_my/fwd/dev/testfwd_ attribute is associated with root attribute _my/other/dev/root_.
_my/other/dev/root_ is polled.
In this case, if _my/fwd/dev_ is ON and _my/other/dev/root_ quality factor switches between VALID and ALARM, the state of _my/fwd/dev_ will not switch correctly to ALARM when _my/other/dev/root_ quality factor is ALARM, unless a client explicitly read _my/fwd/dev/testfwd_ attribute.
This problem is due to the fact that this case is handled by the polling thread for normal attributes, but the polling does not occur in the same device for the forwarded attributes... 
So, in this case, _my/fwd/dev_ should subscribe to _my/other/dev/root_ (it might be already the case, I don't know yet this part of the code) and react when there is a root attribute quality factor change.

bourtemb, 2017-09-13:
(automatically?) Closed by mistake?

Ingvord, 2017-09-13:
Sorry! It was closed accidentally... should be #396  in the comment

commit afe027da (2017-09-13): Resolve #386 conflicts
closed 2022-02-24 by t-b