tango-controls/cppTango #389: cmake installs device_command_info.h in wrong location

jkotan, 2017-08-03:
Hi,

I've just wanted to track  a problem of events which loose their data without any notification https://github.com/tango-controls/pytango/issues/147 in develop branch (i.e. if this problem still exists in the newest version). 
However, when I tried to recompile/install ` cppTango` and `pytango`  I noticed  that `device_command_info.h` is installed in a wrong directory

    -- Installing: /home/jkotan/testtango/include/tango/client/device_command_info.h

instead of

    -- Installing: /home/jkotan/testtango/include/tango/client/device/device_command_info.h

and then when I try to build `pytango` I get

    jkotan@haso228k:~/sources/pytango$ python setup.py  build
    running build
    running build_py
    running build_ext
    building '_tango' extension
    x86_64-linux-gnu-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -fno-strict-aliasing -D_FORTIFY_SOURCE=2 -g -fstack-protector-strong -Wformat -Werror=format-security -fPIC -DPYTANGO_NUMPY_VERSION="1.8.2" -DPYTANGO_HAS_UNIQUE_PTR=1 -I/usr/lib/python2.7/dist-packages/numpy/core/include -I/home/jkotan/testtango/include -I/home/jkotan/sources/pytango/ext -I/home/jkotan/sources/pytango/ext/server -I/usr/include/python2.7 -c /home/jkotan/sources/pytango/ext/api_util.cpp -o build/temp.linux-x86_64-2.7/home/jkotan/sources/pytango/ext/api_util.o -std=c++0x
    In file included from /home/jkotan/testtango/include/tango/client/dbapi.h:32:0,
		     from /home/jkotan/testtango/include/tango.h:93,
		     from /home/jkotan/sources/pytango/ext/api_util.cpp:13:
    /home/jkotan/testtango/include/tango/client/devapi.h:33:53: fatal error: tango/client/device/device_command_info.h: No such file or directory
     #include <tango/client/device/device_command_info.h>
							 ^
while  `device_command_info.h` is in `tango/client/`.

Best regards,
Jan

Ingvord, 2017-08-03:
@jkotan thanks for the report! Looks like our CMake misses proper installation instruction for this file

Ingvord, 2017-08-03:
@jkotan  could you please if this issue is resolved for you in branch cppTango-389?

jkotan, 2017-08-03:
Thanks Igor. Yes, it works.

Ingvord, 2017-08-03:
Resolved in 9e2fc37

closed 2017-08-03 by Ingvord