tango-controls/cppTango #391: Tango::DataReadyEventData::attr_name changed from Tango 8 -> Tango 9

t-b, 2017-09-02:
Transitioning from Tango 8 to Tango 9.2.5a.

For extracting data ready events I've used in tango 8:

```cpp
class EventCallBack : public Tango::CallBack
{
public:

  virtual void push_event(Tango::DataReadyEventData* da)
  {
    std::cout << "Called for: " << da->attr_name << " with counter = " << da->ctr << std::endl;

    try
    {
      auto attr = da->device->read_attribute(da->attr_name);
      std::cout << attr.is_empty() << std::endl;
    }
    catch(Tango::DevFailed &e)
    {
      std::cout << "Caught exception: " << std::endl;
      Tango::Except::print_exception(e);
    }
  }
};
```

which gives

```
Called for: a with counter = 0
0
Called for: a with counter = 1
0
Called for: a with counter = 2
0
```

whereas it breaks with tango 9 as attr_name now holds the full path to the attribute.

```
Called for: tango://thomas-debian-x64:10000/test/myds/i1/a with counter = 6
```

If this behaviour will stay. Is there a stock tango routine for splitting `tango://thomas-debian-x64:10000/test/myds/i1/a` into device part and attribute part?

bourtemb, 2017-09-05:
Hi Thomas,
I think this behaviour will stay like that in order to be able to have a client which subscribes to events coming from different control systems (different TANGO_HOSTs).
As far as I know, there is currently no routine in Tango C++ API to split a fully qualified attribute name into a device part and attribute part, but this work has been done in the HDB++ project for instance.
You can have a look at the following file:
https://github.com/tango-controls/libhdbpp-cassandra/blob/master/src/AttributeName.h
I hope this helps,

t-b, 2017-10-23:
@bourtemb Thanks for the info.

closed 2017-10-23 by t-b