tango-controls/cppTango #392: DeviceAttribute constructor fails for short datatype (Regression from Tango 8)

t-b, 2017-09-02:
The following code 

```
    std::string attr = "myname";
    short data = -111;
    Tango::DeviceAttribute da1;
    da1.set_name(attr);
    da1 << data;
    std::cout << "type: " << da1.get_type() << std::endl;

    Tango::DeviceAttribute da2(attr, data);
    std::cout << "type: " << da2.get_type() << std::endl;
```
gives with tango 9.2.5a

```
type: 100
type: 914331536
```

So the constructor with signed short does not work.

bourtemb, 2017-09-04:
Good catch, Thomas!
Thanks for the bug report.
It seems this bug was introduced when DevEnum support was added in Tango 9.
It looks like DeviceAttribute `data_type` field is not initialized in this case.
The initialization is now necessary for the get_type() method to work as expected.

Ingvord, 2017-09-11:
Resolved in 3721f6e

closed 2017-09-11 by Ingvord

t-b, 2017-09-11:
@bourtemb @Ingvord Thanks.

commit 3721f6e8 (2017-09-11): Fix DeviceAttribute constructor for short and enum data types (#392) (#393)

Close #392
commit b3bad4dd (2017-09-11): Fix DeviceAttribute constructor for short and enum data types (#392) (#393)

Close #392