tango-controls/cppTango #395: Inserting const C string in DeviceData

taurel, 2017-09-11:
Inserting a const C string in a DeviceData instance does not work
(Tango::DeviceData din;
din << "MyString";)

This is due to the inserter method in the DeviceData.h file which is defined for type
char *& or const char *&. The reference must be removed

Ingvord, 2017-09-11:
I would remove support of **char*** from API at all, as C++ way is to use std::string.

bourtemb, 2017-09-11:
When you say removing support for char *, do you mean removing:
`DeviceData::operator << (char *datum)`
and
`DeviceData::operator << (const char *datum)`
or just the first one?

Still I find it convenient to be able to do:
```
DeviceData dd;
dd << "My wonderful string";

```

Ingvord, 2017-09-11:
I mean leave only string. The later should compile due to implicit conversion.

bourtemb, 2017-09-12:
Well, this is what I thought too.
I already tried to keep only the string insertion operator.
It compiles but at run time time, the wrong insertion method is called (operator << (bool) ).

bourtemb, 2017-09-12:
Removing them will also create some compatibility issues.
Tango::DevString is actually a char *
CORBA is also using char * when dealing with DevStrings.
So I think we should not remove them.

Ingvord, 2017-09-12:
Well, I agree - not in Tango 9 LTS ;)

commit 798e4920 (2017-09-13): Tango 9 LTS fix#395 (Inserting const C string in DeviceData) (#396)

Close #395
closed 2017-10-18 by Ingvord