tango-controls/cppTango #397: missing set_source() in the API

jkotan, 2017-09-13:


bourtemb, 2017-09-13:
Hi Jan,

Indeed it seems set_source() method seems to be missing for the groups.
Right now, I think the only work-around seems to be the direct use of DeviceProxy as you wrote or a patch of the library.
I don't know whether this omission was there on purpose... @nleclercq might have memories about that?
Caching is always on by default to improve performances of your control system, especially when you have potentially several clients accessing the same device.
If you don't use caching, then every client which reads an attribute will trigger the execution of the attribute read method (+ always_executed_hook + read_attr_hardware). 
During this execution, if you are using standard Tango serialization model by device or by class, the Tango monitor (a mutex) is taken and no other command or read/write attribute on this device/class will be executed during this time. All the commands and read/write attribute queries will have to wait.
If you have many clients, unless you are using caching, the system will become slow and the commands and attribute read/write queries from the clients will take a longer time to execute, until you eventually reach a point where you will get timeout exceptions (Not able to acquire serialization errors, actually).

jkotan, 2017-09-13:
Thanks for your answer. So it seems Group class is useless, at least for data acquisition. 
Do you think it is possible to add `set_source()` into (Py)Tango 9 as well?

andygotz, 2017-09-13:
I don't think Groups are useless for data acquisition. At the ESRF we use them for acquiring many BPMs. The advantage of Group calls is all requests are fired off asynchronously and then the client waits for all of them to answer within the timeout. This is useful for reading a large number of devices on many servers and hosts.

Caching is probably not implemented because it was not clear at what level to cache the value. Group level or individual device level. If the Device is polled then the Group call automatically returns with the cached value (unless the proxy selects source=DEV).

We should discuss what happens when you do set_source in the Group call before implementing it. If we put it at the client level who will do the polling?

nleclercq, 2017-09-13:
Sorry guys, I missed this discussion. 
Groups are definitively useful in many cases except for the one @jkotan wants to use them. However, adding a `set_source`to the Group class is feasible (1)  and I don't see anything that could prevent use from propagating it to pytango.  Just a question of time and/or resources.

(1) in fact anything available in the DeviceProxy interface can be exposed in the Group interface.

jkotan, 2017-09-14:
Thank you for your answer. I've just try it  and it works perfectly fine in C++.
Unfortunately, in PyTango the following code:

    grp = PyTango.Group("motors")
    for i in range(1, 17):
        grp.add("p09/motor/exp.%02i" % i)
    for i in range(1, 17):
        dp = grp.get_device("p09/motor/exp.%02i" % i)
        dp.set_source(PyTango.DevSource.DEV)
        print dp.get_source()
        dp = grp.get_device("p09/motor/exp.%02i" % i)
        print dp.get_source()

fails 

    DEV
    CACHE_DEV
    DEV
    CACHE_DEV
    DEV
    CACHE_DEV
    DEV
    CACHE_DEV
    DEV
    CACHE_DEV
    DEV
    CACHE_DEV
    DEV
    CACHE_DEV
    DEV
    CACHE_DEV
    DEV
    CACHE_DEV
    DEV
    CACHE_DEV
    DEV
    CACHE_DEV
    DEV
    CACHE_DEV
    DEV
    CACHE_DEV
    DEV
    CACHE_DEV
    DEV
    CACHE_DEV
    DEV
    CACHE_DEV

So it looks like it is a bug of PyTango. I will create a ticket.

Bests,
Jan

jkotan, 2017-09-14:
Since @nleclercq solution is quite compact I think we can close the issue.
Thank you.

closed 2017-09-14 by jkotan