tango-controls/cppTango #399: DeviceProxy subscribe_event performance

guifrecuni, 2017-09-19:
Hi, debugging performance issues with some graphical applications with many (500+) attributes, we have found a performance regression between TANGO7 and TANGO9.

We were used to ~60us for each subscribe_event in C++ TANGO 7.2.6
and now we experience ~10ms for each subscribe_event in C++ TANGO 9.2.5

Please, find attached the test code I have used:

`
$ ./subscribe_loop_t7 sys/tg_test/1 double_scalar
avg subscription takes 5.8716e-05 seconds

$ ./subscribe_loop_t9 sys/tg_test/1 double_scalar
avg subscription takes 0.0104682 seconds
`

Could somebody confirm this?

Thanks,

Guifré.
PS: github did not support .cpp file, renamed to .txt
[subscribe_loop.cpp.txt](https://github.com/tango-controls/cppTango/files/1313595/subscribe_loop.cpp.txt)

bourtemb, 2017-09-19:
Hi Guifré,

Thanks for reporting this issue.

As I explained on the forum, a quick look at the source code led me to the following lines:
https://github.com/tango-controls/cppTango/blob/tango-9-lts/cppapi/client/event.cpp#L3113

There is indeed a sleep of 10ms (even 25ms on Windows!) during the subscription process and as explained in the source code:

> This is required in case there is a push_event in the read_attribute (or pipe) method on the device side. This sleep gives time to ZMQ to send its subscription message

Maybe we could improved this but it looks tricky at first sight because it looks like this sleep seems to be there because not having it would generate problems in some specific cases.
Deeper investigations and tests will be needed in order to better understand the problems generated when the sleep is not there.

Cheers,
Reynald

bourtemb, 2017-09-20:
I ran the unit tests after removing the 10ms sleep on my fork.
One test didn't pass. 
https://travis-ci.org/bourtemb/cppTango/builds/277251702
So it looks like a unit test was added on purpose to test the case where a device server pushes an event in the read_pipe method. It seems it is expected to have the callback called twice during the subscription phase in this case.
I think this could be useful only if the device server is sending an event with a value and/or timestamp different than the value returned by read_attribute/read_pipe method. In any case, the callback is called at least once after the synchronous call occurring during the subscription.
The 10ms might be useful during the subscription to a user event and if the read_attribute/read_pipe method could generate a user event.
I have difficulties to imagine another use case where it would be interesting to call the callback twice during the subscription when there is a push_event in the read_attribute/read_pipe device server method.
It does not sound healthy to me to push an event in read_attribute/read_pipe device server method with a value different than the value returned by this read_attribute/read_pipe method...

I tend to think we should do a special case for user event subscription but not for the other cases.
I you feel lucky, you could try to remove this 10ms sleep and monitor/report any unexpected behaviour ;-)...

Kind regards,
Reynald

andygotz, 2017-09-20:
Thanks for looking into this. I agree that it is worth trying to remove the 10ms and seeing if the special case it is trying to avoid happens. A sleep is always a touchy solution. Why 10ms and not more or less? 

From your description it seems like the issue arises because we request an event to be sent when subscribing to simulate a synchronous read of the attribute. Maybe adding a call to subscribe without reading the attribute could be useful. I know this has been asked for in the past.

Andy

bourtemb, 2017-09-20:
I think this is another issue.
I didn't write the issue is there because of that.  I even wanted to say that the issue should not be there thanks to that, unless the device server programmer is doing weird things...

guifrecuni, 2017-09-20:
Thank you both, we will keep you updated with our progress.

Thanks,

Guifré.

andygotz, 2017-10-25:
Hi Guifre, 
now that ICALEPCS is over have you any news for us on this topic? We can discuss it at the kernel meeting if you prefer. 
Cheers
Andy

sergirubio, 2017-10-27:
I did some tries reducing the sleep to 1 ms and I got several API event timeout that ended in segfault. So it seems that there's a problem behind.

Just one poing. I'm currently running a Tango9 client on Tango7 devices, so the events are notifd and not ZMQ, so why am I affected by this sleep issue !?!?!

Sergi

bourtemb, 2017-11-02:
Hi Sergi,

The sleep is in a part of the code which seems to be executed for both ZMQ events and notifd events. This is why you are also affected when using notifd events.
The sleep is in a method named EventConsumer::get_fire_sync_event(). The EventConsumer class is used when using both implementations of events.

Did you get the chance to investigate the seg faults? Any core file?
Is it easy to reproduce?
Kind regards,
Reynald

bourtemb, 2018-01-26:
This issue should be fixed with PR #415 .
The same needs to be done in Tango 10 (master)  branch.

bourtemb, 2019-03-29:
Solved with #415 but with a side effect in some situations as explained in #524

closed 2019-03-29 by bourtemb