tango-controls/cppTango #408: Cmake build broken

t-b, 2017-11-02:
Doing

```
mkdir build
cd build
cmake ..
```

results in

```
Remove tango.h, tangoSK.cpp and tangoDynSK.cpp from idl if exist
rm: das Entfernen von 'tango.h' ist nicht möglich: Datei oder Verzeichnis nicht gefunden
rm: das Entfernen von 'tangoSK.cpp' ist nicht möglich: Datei oder Verzeichnis nicht gefunden
rm: das Entfernen von 'tangoDynSK.cpp' ist nicht möglich: Datei oder Verzeichnis nicht gefunden
Generate tango.h, tangoSK.cpp and tangoDynSK.cpp from idl
Using OMNIIDL_PATH=
Using IDL=/usr/local/include
omniidl: '/usr/local/include/tango.idl' does not exist
CMake Error at cppapi/server/idl/CMakeLists.txt:16 (message):
   Failed to generate source files from idl. rv=1

Applying enhancement /home/firma/devel/cppTango/cppapi/server/idl/EnhanceAttributeValue_4
sed: tango.h kann nicht gelesen werden: Datei oder Verzeichnis nicht gefunden
CMake Error at cppapi/server/idl/CMakeLists.txt:29 (message):
   Failed to apply /home/firma/devel/cppTango/cppapi/server/idl/EnhanceAttributeValue_4. rv=2

Applying enhancement /home/firma/devel/cppTango/cppapi/server/idl/EnhanceAttributeValue_5
sed: tango.h kann nicht gelesen werden: Datei oder Verzeichnis nicht gefunden
CMake Error at cppapi/server/idl/CMakeLists.txt:29 (message):
   Failed to apply /home/firma/devel/cppTango/cppapi/server/idl/EnhanceAttributeValue_5. rv=2

Applying enhancement /home/firma/devel/cppTango/cppapi/server/idl/EnhanceDevPipeData
sed: tango.h kann nicht gelesen werden: Datei oder Verzeichnis nicht gefunden
CMake Error at cppapi/server/idl/CMakeLists.txt:29 (message):
   Failed to apply /home/firma/devel/cppTango/cppapi/server/idl/EnhanceDevPipeData. rv=2

HOST_NAME=thomas-debian-x64
```

On both master and tango9-lts branch.

bourtemb, 2017-11-02:
Hi Thomas,

tango.h, tangoSK.cpp and tangoDynSK.cpp files are generated during the CMake configuration process.
For this to work, you need to install tango-idl first (coming from https://github.com/tango-controls/tango-idl repository).
tango-idl is listed in the prerequisites in Install.md file (https://github.com/tango-controls/cppTango/blob/master/INSTALL.md)
You also need to make sure omniidl is installed on your system because it used to generate the above files from the IDL file.

t-b, 2017-11-02:
@bourtemb Thanks for the hints. I've changed the title accordingly as cmake should complain.

closed 2017-11-02 by Ingvord

t-b, 2017-11-03:
@Ingvord Thanks.

commit 277d9880 (2017-11-02): Close #408