tango-controls/cppTango #410: Appveyor build fails

Ingvord, 2017-11-02:
```
CMake Error at C:/Program Files (x86)/CMake/share/cmake-3.9/Modules/FindPkgConfig.cmake:640 (message):
  None of the required 'tangoidl' found
Call Stack (most recent call first):
  configure/CMakeLists.txt:19 (pkg_search_module)
  CMakeLists.txt:18 (include)
-- Checking for one of the modules 'omniORB4'
CMake Error at C:/Program Files (x86)/CMake/share/cmake-3.9/Modules/FindPkgConfig.cmake:640 (message):
  None of the required 'omniORB4' found
Call Stack (most recent call first):
  configure/CMakeLists.txt:43 (pkg_search_module)
  CMakeLists.txt:18 (include)
-- Checking for one of the modules 'omniDynamic4'
CMake Error at C:/Program Files (x86)/CMake/share/cmake-3.9/Modules/FindPkgConfig.cmake:640 (message):
  None of the required 'omniDynamic4' found
Call Stack (most recent call first):
  configure/CMakeLists.txt:45 (pkg_search_module)
  CMakeLists.txt:18 (include)
FATALomniORB4 library is required for Tango
-- Checking for one of the modules 'libzmq'
CMake Error at C:/Program Files (x86)/CMake/share/cmake-3.9/Modules/FindPkgConfig.cmake:640 (message):
  None of the required 'libzmq' found
Call Stack (most recent call first):
  configure/CMakeLists.txt:91 (pkg_search_module)
  CMakeLists.txt:18 (include)
```

aureocarneiro, 2019-04-12:
I have the same error !

bourtemb, 2019-04-23:
Hi @aureocarneiro 

Are you using the master branch or the tango-9-lts branch?
appveyor should work on the tango-9-lts branch, which is the officially supported branch.
Are you trying to build manually cppTango library on Windows?

aureocarneiro, 2019-04-24:
Hi @bourtemb 

It was just my mistake, it's all okay now. Thank you!

closed 2019-06-05 by t-b