tango-controls/cppTango #426: Silent/ Quiet install/ uninstall

lzytniak, 2018-01-11:
@NexeyaSGara How to silent install/uninstall TangoSetup-9.2.2_XXX.exe?
How to set e.g. InstallDir from command line?

NexeyaSGara, 2018-01-11:
@lzytniak since it's a classical msi, you can use the classical way.
Install
msiexec /i libtango*.msi /q+ /passive 

Uninstall
msiexec /x libtango*.msi /q+ /passive

for specifying your path you must add INSTALL_ROOT="d:\my_new_path"

bourtemb, 2018-01-31:
It looks like the solution proposed by @NexeyaSGara is useful when using the artifacts built from appveyor.
The question was about TangoSetup-9.2.2_XXX.exe which is older than that. It is an exe file, not an msi file.
@lzytniak, did the previous answer helped you to solve your issue or do you need an answer specifically for Tango 9.2.2 exe file?
Do you still need some help on this topic? If not, we will close this issue.
Kind regards,
Reynald

lzytniak, 2018-02-02:
Yes, it works! Thank you @NexeyaSGara 
[image]

I think it would be good to add in the documentation section like: "Installing Without UI"
https://docs.python.org/3/using/windows.html#installing-without-ui
@bourtemb what do you think?

bourtemb, 2018-02-05:
> I think it would be good to add in the documentation section like: "Installing Without UI"
https://docs.python.org/3/using/windows.html#installing-without-ui
@bourtemb what do you think?

I agree with you. It would be good to add something like that in the doc. 👍

bourtemb, 2018-02-05:
> Is it possible to install Fully silent mode?
@lzytniak, the examples you provided on your screenshot are with TangoSetup-9.2.2_win64.exe which is not an msi file so it is simply ignoring the options you are passing.

bourtemb, 2019-07-17:
I created https://github.com/tango-controls/tango-doc/issues/286 issue in tango-doc repository to keep track of the new documentation suggestion.
I close this issue in cppTango since this is more a documentation related issue.
@lzytniak , please re-open if you need more help, or please comment https://github.com/tango-controls/tango-doc/issues/286.

closed 2019-07-17 by bourtemb