tango-controls/cppTango #428: Regular test failure in new_tests/cxx_fwd_att.cpp:749

bourtemb, 2018-01-18:
There is quite often a failure of the Travis CI tests due to an assertion failure (Assertion failed: cb.cb_err == 1) in new_tests/cxx_fwd_att.cpp file at line 749

Here is an example of logs seen on Travis:

> 1: Testing Fwd Att :
> 1: 
> 1: 	error case in forwarded attribute configuration --> OK
> 1: 	reading forwarded attribute --> OK
> 1: 	writing forwarded attribute --> OK
> 1: 	data propagation --> OK
> 1: 	tango://172.17.0.3:10000/test/debian8/10/short_attr_rw==tango://172.17.0.3:10000/test/debian8/10/short_attr_rw
> 1: 	attribute configuration --> OK
> 1: 	attribute configuration propagation --> OK
> 1: 	label is local --> OK
> 1: 	write read on forwarded attribute --> OK
> 1: 	polling on a forwarded attribute --> OK
> 1: Tango exception
> 1: Severity = ERROR 
> 1: Error reason = API_AttributePollingNotStarted
> 1: Desc : The polling (necessary to send events) for the attribute short_attr_rw is not started
> 1: Origin : DServer::event_subscription
> 1: 
> 1: Tango exception
> 1: Severity = ERROR 
> 1: Error reason = API_CommandFailed
> 1: Desc : Failed to execute command_inout on device dserver/devtest/test, command ZmqEventSubscriptionChange
> 1: Origin : Connection::command_inout()
> 1: 
> 1: Tango exception
> 1: Severity = ERROR 
> 1: Error reason = API_DSFailedRegisteringEvent
> 1: Desc : Device server send exception while trying to register event
> 1: Origin : EventConsumer::connect_event()
> 1: 
> 1: Tango exception
> 1: Severity = ERROR 
> 1: Error reason = API_CommandFailed
> 1: Desc : Failed to execute command_inout on device dserver/fwdtest/test, command ZmqEventSubscriptionChange
> 1: Origin : Connection::command_inout()
> 1: 
> 1: Tango exception
> 1: Severity = ERROR 
> 1: Error reason = API_DSFailedRegisteringEvent
> 1: Desc : Device server send exception while trying to register event
> 1: Origin : EventConsumer::connect_event()
> 1: 
> 1: 		EventCallBack::push_event(): called attribute tango://172.17.0.3:10000/test/fwd_debian8/10/fwd_short_rw event periodic
> 1: 		EventCallBack::push_event(): called attribute tango://172.17.0.3:10000/test/fwd_debian8/10/fwd_short_rw event periodic
> 1: 		EventCallBack::push_event(): called attribute tango://172.17.0.3:10000/test/fwd_debian8/10/fwd_short_rw event periodic
> 1: 		EventCallBack::push_event(): called attribute tango://172.17.0.3:10000/test/fwd_debian8/10/fwd_short_rw event periodic
> 1: 
> 1: In FwdAttTestSuite::test_event_on_a_forwarded_attribute:
> 1: /home/tango/src/cpp_test_suite/cxxtest/include/../../new_tests/cxx_fwd_att.cpp:749: Error: Assertion failed: cb.cb_err == 1
> 1: TEST FAILED in:
> 1: 1 iteration of test case
> 1: 1 iteration of test suite
> 1: 
>  1/57 Test  -1: CXX::runner ......................***Failed  146.05 sec

Ingvord, 2018-01-29:
Same as #384

closed 2018-01-29 by Ingvord

bourtemb, 2018-01-29:
Ooops, sorry for creating a duplicate...