tango-controls/cppTango #433: Crashes in data ready event callback

t-b, 2018-02-09:
In the repository, https://github.com/t-b/crash-tango-data-ready-event, I have a server and a client.

Steps to reproduce:
- Compile both
- Declare Test DS as Test/ts/1
- Start DS
- Start Client with `make && gdb -ex run -args ./test-client`

Here, with tango 9.2.5a on debian stretch I get a crash after ~750k events.

gdb backtrace:
```
DRE: m_counter: 1434177824, e->ctr: 743725

Thread 4 "test-client" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fffee537700 (LWP 5895)]
0x00007ffff7853016 in Tango::ZmqEventConsumer::push_zmq_event (this=this@entry=0x5555557bd120, ev_name="tango://thomas:10000/test/ts/1/datasync.data_ready",
    endian=endian@entry=1 '\001', event_data=..., error=<optimized out>, ds_ctr=<optimized out>) at zmqeventconsumer.cpp:2909
2909                                        callback->push_event(event_data_);
(gdb) bt
#0  0x00007ffff7853016 in Tango::ZmqEventConsumer::push_zmq_event (this=this@entry=0x5555557bd120, ev_name="tango://thomas:10000/test/ts/1/datasync.data_ready",
    endian=endian@entry=1 '\001', event_data=..., error=<optimized out>, ds_ctr=<optimized out>) at zmqeventconsumer.cpp:2909
#1  0x00007ffff785730f in Tango::ZmqEventConsumer::process_event (this=this@entry=0x5555557bd120, received_event_name=..., received_endian=..., received_call=...,
    event_data=...) at zmqeventconsumer.cpp:594
#2  0x00007ffff7858257 in Tango::ZmqEventConsumer::run_undetached (this=0x5555557bd120, arg=<optimized out>) at zmqeventconsumer.cpp:320
#3  0x00007ffff5eb66c1 in omni_thread_wrapper () from /usr/lib/libomnithread.so.3
#4  0x00007ffff48b8494 in start_thread (arg=0x7fffee537700) at pthread_create.c:333
#5  0x00007ffff3d5daff in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:97
(gdb) info locals
event_data_ = 0x7fffe80168d0
callback = 0x7fffffffdb40
ev_queue = 0x0
_mon = {mon = 0x5555557c86c0, auto_self = {_self = <optimized out>, _dummy = 0}}
esspos = {ev_queue = 0x0, callback = 0x7fffffffdb40, id = 1}
cb_nb = <optimized out>
cb_ctr = 1
attr_value = <optimized out>
z_attr_value_5 = <optimized out>
ev_attr_ready = true
att_name = "datasync"
data_type = <optimized out>
ev_attr_conf = false
pipe_event = false
event_name = "data_ready"
vers = 0
no_unmarshalling = false
missed_event_data = 0x0
missed_ready_event_data = 0x0
missed_dev_pipe_data = 0x0
attr_value_3 = <optimized out>
attr_conf_2 = <optimized out>
att_ready = 0x5555557b7f10
dev_intr_change = 0x0
missed_event = <optimized out>
pos = <optimized out>
full_att_name = "tango://thomas:10000/test/ts/1/datasync"
missed_conf_event_data = 0x0
missed_dev_intr_event_data = 0x0
attr_conf_3 = <optimized out>
err_missed_event = false
dev_pipe = 0x0
ev_dev_intr = false
dev_attr = 0x0
z_attr_value_4 = <optimized out>
attr_conf_5 = 0x0
errors = {<_CORBA_Unbounded_Sequence<Tango::DevError>> = {<_CORBA_Sequence<Tango::DevError>> = {pd_max = 0, pd_len = 0, pd_rel = true, pd_bounded = false,
      pd_buf = 0x0}, <No data fields>}, <No data fields>}
attr_info_ex = 0x0
pos1 = <optimized out>
---Type <return> to continue, or q <return> to quit---
new_tango_host = "tango://thomas:10000/test/ts/1/datasync.data_ready"
map_lock = true
ipos = <optimized out>
loop = <optimized out>
no_db_dev = <optimized out>
first_search_succeed = true
pos = <optimized out>
base_tango_host = "tango://thomas:10000/"
canon_ev_name = "test/ts/1/datasync.data_ready"

```

bourtemb, 2018-02-28:
It looks like the crash is occurring when the main method of your client is exiting.
When it exits, it will delete the CallBack object named cb.
The crash occurs because the Tango::DeviceProxy object named device is not deleted as it should be at the end of the main() method so the Tango library is still trying to process the DataReady events received which are in still in the queue (which should be containing many events, probably about 250000 in your case), even after the 5 seconds sleep time, because there is no sleep time in the GenerateDataSync loop. When the Tango library receives a DataReady event, it tries to execute CallBack::push_event() on cb object which has just been deleted at the end of the main method.
The solution is to delete the DeviceProxy object before exiting from main().
When the DeviceProxy is deleted, it will unsubscribe all the event which were subscribed via this DeviceProxy object, so the callback will not be called any longer.

t-b, 2018-03-03:
@bourtemb Thanks for the explanation. Much appreciated. I now do see my error.

closed 2018-03-03 by t-b