tango-controls/cppTango #439: Memory Leak on get_device_property() method

FPO28, 2018-02-27:
Hi everyone,

We have C++ device servers, and while testing polling on init() method, I observed a memory leak from get_device_property() of about 3.9kb by call. I saw the memory leak on two device servers generated by POGO.  

To observe it, we can just call "get_device_property();" in "always_executed_hook();" method (+2MB in about 6 minutes). I tested it on several device servers, and the memory leak seems to be the same on each.
  
For information, the leak was observed on Windows 7 and on Windows Server 2016.

I searched on documentation and on the forum, and didn't find any result. Is this known? Was it solved on Tango 9?

I currenty use Tango 8.1.2, would it be possible to just get the modification and still use Tango 8.1.2 ?

I'm not sure it's on the most appropriate topic, I will move it if this is the case. 
Thanks for you answers.

Best regards
Florian POURCHAYRE

bourtemb, 2018-02-27:
Hi Florian,

Thank you very much for the bug report.
Could you please attach the source code of a very basic device server reproducing the problem?
What version of POGO did you use?
What kind of device property (type) are you trying to read?

Cheers,
Reynald

FPO28, 2018-02-27:
Hi Reynald, 

Thank you for your quick answer.

I created a new device server with 3 properties and one command.
 I joined a zip file containing : 
- A XMI file, 
- All ressources generated by POGO, including a .sln to generate app.

I tested it, and see again the memory increasing just by calling get_device_property() in always_executed_hook().

When I tested my device servers, the properties were scalar double, strings, and int.

I use device server generated by Pogo 8.1.14 and by Pogo6.

Thanks again,

Florian Pourchayre

[TestDeviceServer.zip](https://github.com/tango-controls/cppTango/files/1763389/TestDeviceServer.zip)

bourtemb, 2018-02-28:
> since you are on Windows, it might as well come from the suspicious line which is (still) not executed on Windows in DbDatum destructor, see ~DbDatum::~DbDatum() in cppapi/client/dbapi_datum.cpp file.

This is actually a problem only if the code was compiled with MSVC++ < 10 (Before Visual Studio 2010).
Recent MS VC++ version have support for unique pointers so this delete is not necessary since Visual Studio 2010.

FPO28, 2018-03-01:
Hi Reynald, 

Thank you for your help !

Indeed, we use Visual Studio 2008 to compile the code.

I will look more into it! :)

Regards,
Florian Pourchayre

t-b, 2019-06-05:
Fixed via #488. If not please reopen.

closed 2019-06-05 by t-b