tango-controls/cppTango #443: push_archive_event() or push_change_event() methods causing a crash in some specific situations

bourtemb, 2018-03-27:
We have observed some device servers crashes when _DeviceImpl::push_archive_event()_ or _Device::Impl::push_change_event()_ methods are invoked in some specific situations.

For instance when these methods are invoked after having invoked attr.set_value() inside a _read_MyAttributeName(Tango::Attribute &attr)_ method.
Here is a code example for an attribute named _Value_:

```
void TestArchive::read_Value(Tango::Attribute &attr)
{
    DEBUG_STREAM << "TestArchive::read_Value(Tango::Attribute &attr) entering... " << endl;
    /*----- PROTECTED REGION ID(TestArchive::read_Value) ENABLED START -----*/
    Tango::DevLong64 x = 0x5555333377771111;
    *attr_Value_read = x;
    // No problem when push_<archive|change>_event is invoked before attr.set_value()
    //push_archive_event(attr.get_name(), &x);
    //	Set the attribute value
    attr.set_value(attr_Value_read);
    push_archive_event(attr.get_name(),&x); // A crash will occur after exiting TestArchive::read_Value
    /*----- PROTECTED REGION END -----*/	//	TestArchive::read_Value
}

```

A similar problem could arise when reading an attribute _X_ is triggering push_archive_event() or push_change_event() on another attribute _Y_ and if these 2 attributes are read together in the same read_attributes(). In this particular case, a crash will occur only if _Y_ is read before _X_.
This case may happen in Tango 9 when the 2 attributes are polled at the same polling period, since Tango 9 is using read_attributes() by default to read all the attributes configured with the same polling period.

stanislaw55, 2018-09-05:
Hello, 
I think I've encountered this issue while using python Facade DS. Could you confirm?  Here's my backtrace:
```#0  0x00007f7a7260a15f in Tango::AttrValUnion::operator>>=(cdrStream&) const () from /lib64/libtango.so.9
#1  0x00007f7a7260a6b5 in Tango::AttributeValue_4::operator>>=(cdrStream&) const () from /lib64/libtango.so.9
#2  0x00007f7a72602f2d in Tango::ZmqEventSupplier::push_event(Tango::DeviceImpl*, std::string, std::vector<std::string, std::allocator<std::string> >&, std::vector<double, std::allocator<double> >&, std::vector<std::string, std::allocator<std::string> >&, std::vector<long, std::allocator<long> >&, Tango::EventSupplier::SuppliedEventData&, std::string&, Tango::DevFailed*, bool) () from /lib64/libtango.so.9
#3  0x00007f7a725fda10 in Tango::ZmqEventSupplier::push_event_loop(Tango::DeviceImpl*, Tango::EventType, std::vector<std::string, std::allocator<std::string> >&, std::vector<double, std::allocator<double> >&, std::vector<std::string, std::allocator<std::string> >&, std::vector<long, std::allocator<long> >&, Tango::EventSupplier::SuppliedEventData&, Tango::Attribute&, Tango::DevFailed*) () from /lib64/libtango.so.9
#4  0x00007f7a7241f92f in Tango::Attribute::fire_change_event(Tango::DevFailed*) () from /lib64/libtango.so.9
#5  0x00007f7a72f95cd6 in PyDeviceImpl::push_change_event(Tango::DeviceImpl&, boost::python::str&, boost::python::api::object&, double, Tango::AttrQuality) ()
   from /usr/lib64/python2.7/site-packages/tango/_tango.so
#6  0x00007f7a72f8b364 in boost::python::objects::caller_py_function_impl<boost::python::detail::caller<void (*)(Tango::DeviceImpl&, boost::python::str&, boost::python::api::object&, double, Tango::AttrQuality), boost::python::default_call_policies, boost::mpl::vector6<void, Tango::DeviceImpl&, boost::python::str&, boost::python::api::object&, double, Tango::AttrQuality> > >::operator()(_object*, _object*) ()
   from /usr/lib64/python2.7/site-packages/tango/_tango.so
#7  0x00007f7a71b4f29c in boost::python::objects::function::call(_object*, _object*) const () from /lib64/libboost_python.so.1.53.0
#8  0x00007f7a71b4f5f8 in boost::detail::function::void_function_ref_invoker0<boost::python::objects::(anonymous namespace)::bind_return, void>::invoke(boost::detail::function::function_buffer&) ()
   from /lib64/libboost_python.so.1.53.0
#9  0x00007f7a71b580f3 in boost::python::detail::exception_handler::operator()(boost::function0<void> const&) const () from /lib64/libboost_python.so.1.53.0
#10 0x00007f7a72eb80d3 in boost::detail::function::function_obj_invoker2<boost::_bi::bind_t<bool, boost::python::detail::translate_exception<Tango::NotAllowed, void (*)(Tango::NotAllowed const&)>, boost::_bi::list3<boost::arg<1>, boost::arg<2>, boost::_bi::value<void (*)(Tango::NotAllowed const&)> > >, bool, boost::python::detail::exception_handler const&, boost::function0<void> const&>::invoke(boost::detail::function::function_buffer&, boost::python::detail::exception_handler const&, boost::function0<void> const&) () from /usr/lib64/python2.7/site-packages/tango/_tango.so
#11 0x00007f7a71b580c8 in boost::python::detail::exception_handler::operator()(boost::function0<void> const&) const () from /lib64/libboost_python.so.1.53.0
#12 0x00007f7a72eb8083 in boost::detail::function::function_obj_invoker2<boost::_bi::bind_t<bool, boost::python::detail::translate_exception<Tango::DeviceUnlocked, void (*)(Tango::DeviceUnlocked const&)>, boost::_bi::list3<boost::arg<1>, boost::arg<2>, boost::_bi::value<void (*)(Tango::DeviceUnlocked const&)> > >, bool, boost::python::detail::exception_handler const&, boost::function0<void> const&>::invoke(boost::detail::function::function_buffer&, boost::python::detail::exception_handler const&, boost::function0<void> const&) () from /usr/lib64/python2.7/site-packages/tango/_tango.so
#13 0x00007f7a71b580c8 in boost::python::detail::exception_handler::operator()(boost::function0<void> const&) const () from /lib64/libboost_python.so.1.53.0
#14 0x00007f7a72eb8033 in boost::detail::function::function_obj_invoker2<boost::_bi::bind_t<bool, boost::python::detail::translate_exception<Tango::EventSystemFailed, void (*)(Tango::EventSystemFailed const&)>, boost::_bi::list3<boost::arg<1>, boost::arg<2>, boost::_bi::value<void (*)(Tango::EventSystemFailed const&)> > >, bool, boost::python::detail::exception_handler const&, boost::function0<void> const&>::invoke(boost::detail::function::function_buffer&, boost::python::detail::exception_handler const&, boost::function0<void> const&) () from /usr/lib64/python2.7/site-packages/tango/_tango.so
#15 0x00007f7a71b580c8 in boost::python::detail::exception_handler::operator()(boost::function0<void> const&) const () from /lib64/libboost_python.so.1.53.0
#16 0x00007f7a72eb7fe3 in boost::detail::function::function_obj_invoker2<boost::_bi::bind_t<bool, boost::python::detail::translate_exception<Tango::AsynReplyNotArrived, void (*)(Tango::AsynReplyNotArrived const&)>, boost::_bi::list3<boost::arg<1>, boost::arg<2>, boost::_bi::value<void (*)(Tango::AsynReplyNotArrived const&)> > >, bool, boost::python::detail::exception_handler const&, boost::function0<void> const&>::invoke(boost::detail::function::function_buffer&, boost::python::detail::exception_handler const&, boost::function0<void> const&) () from /usr/lib64/python2.7/site-packages/tango/_tango.so
#17 0x00007f7a71b580c8 in boost::python::detail::exception_handler::operator()(boost::function0<void> const&) const () from /lib64/libboost_python.so.1.53.0
[... cut ...]
#104 0x00007f7a723f2a14 in Tango::ZmqEventConsumer::process_event(zmq::message_t&, zmq::message_t&, zmq::message_t&, zmq::message_t&) () from /lib64/libtango.so.9
#105 0x00007f7a723f3b2e in Tango::ZmqEventConsumer::run_undetached(void*) () from /lib64/libtango.so.9
#106 0x00007f7a703fe67e in omni_thread_wrapper () from /lib64/libomnithread.so.4
#107 0x00007f7a85c5fe25 in start_thread (arg=0x7f7a5f7fe700) at pthread_create.c:308
#108 0x00007f7a85280bad in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:113```

bourtemb, 2018-09-05:
Hi,

Difficult to say with only this backtrace... Can you reproduce it easily?
Where can we get the code of this Facade python DS? I didn't find it on the [Classes Catalog](http://www.tango-controls.org/developers/dsc/search/?search=facade).
Do you think the code does one of the things I have described in the description of this issue?

Kind regards,
Reynald

stanislaw55, 2018-09-06:
Hi @bourtemb 

> Can you reproduce it easily?

Unfortunately, no. It's happening just from time to time, especially in a week after shutdown. This Facade DS is gathering informations about beam, lifetime, dose, current rate etc. 

> Do you think the code does one of the things I have described in the description of this issue?

Few attributes depend on each other so yes, I think it _could_ be the issue you described, since `facadedevice` _could_ read two or more attributes in one row.

After some investigation, I've found [that ](https://github.com/tango-controls/pytango/issues/123) issue. Segmentation fault happens during push event of attribute with `AttrQuality` set to `INVALID`.  Do you think those two issues are correlated?

bourtemb, 2018-09-06:
Hi, do you know what is the cppTango version used underneath in your case?
Your problem looks similar to this issue which was fixed in Tango 9.2.5 :+1: https://github.com/tango-controls/cppTango/issues/246

stanislaw55, 2018-09-06:
We are using Tango version 9.2.5a. I've launched the example case from issue you linked, and it's working, no segfaults. So it has to be something else which causes segfualts, or my situation is more specific and fix yo provided in 9.2.5 doe not apply. What do you think? Should I give more detials? I can send you code of my Facade DS by email.

bourtemb, 2018-09-06:
Hi,
You can attach the code as a comment (attach a zip file) on Github with indications on how to configure and use the device server to reproduce the problem.
Of course, if you could isolate the problem in a C++ device server, we would prefer and it would take us much less time to debug this issue.
Kind regards,
Reynald

stanislaw55, 2018-09-10:
I'll try to reproduce this bug in C++ layer and then attach example. In meantime, I've got another idea: could be possible that this kind of error occurs when trying to push archive event?

mliszcz, 2019-12-19:
Hi @bourtemb 

I've checked the crash from the original post. The problem is that `push_archive_event` postcondition is to delete underlying CORBA sequence:
https://github.com/tango-controls/cppTango/blob/c34b0496626e9625484fffcfb9c14d2d58b0ba45/cppapi/server/attribute.cpp#L4294-L4296

When attribute read callback exits and attribute's quality is valid and "value flag" is set, Tango assumes that sequence was allocated (as set_value does) and tries to send the sequence over the network, generating a crash (because at the very end of read callback we deleted the sequence by pushing the event):

Gdb:
```
Thread 9 "CppTango443" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fb405769700 (LWP 1436)]
Tango::DeviceImpl::data_into_net_object (this=this@entry=0x5598cbd3c730, att=..., aid=..., index=index@entry=0, w_type=w_type@entry=Tango::READ, del_seq=del_seq@entry=true) at /home/michal/Documents/cppTango/cppapi/server/device.cpp:5189
5189                DATA_IN_OBJECT(DevVarLong64Array, get_long64_value, dummy_long64_att_value, long64_att_value);
+(gdb) bt
#0  Tango::DeviceImpl::data_into_net_object (this=this@entry=0x5598cbd3c730, att=..., aid=..., index=index@entry=0, w_type=w_type@entry=Tango::READ, del_seq=del_seq@entry=true) at /home/michal/Documents/cppTango/cppapi/server/device.cpp:5189
```

Valgrind:
```
==3191== Invalid read of size 1
==3191==    at 0x51E7625: release (seqTemplatedecls.h:93)
==3191==    by 0x51E7625: Tango::DeviceImpl::data_into_net_object(Tango::Attribute&, Tango::_AttributeIdlData&, long, Tango::AttrWriteType, bool) (device.cpp:5195)
==3191==    by 0x521358F: Tango::Device_3Impl::read_attributes_no_except(Tango::DevVarStringArray const&, Tango::_AttributeIdlData&, bool, std::vector<long, std::allocator<long> >&) (device_3.cpp:909)
==3191==    by 0x5229205: Tango::Device_5Impl::read_attributes_5(Tango::DevVarStringArray const&, Tango::DevSource, Tango::ClntIdent const&) (device_5.cpp:359)
```

You can get a similar crash with following read callback:
```cpp
void CppTango443::read_value(Tango::Attribute &attr)
{
    Tango::DevLong64 x = 0x5555333377771111;
    *attr_value_read = x;
    /* attr.set_value(attr_value_read); // commented-out! */
    attr.set_value_flag(true);
    attr.set_quality(Tango::ATTR_VALID);
}
```

Now, the solution. I think we can:
1. Write in the manual / API docs that any events must be pushed before calling attr.set_value,
2. Add new parameter to push_event / new method for pushing event, that will not release the sequence - not tested but should work.
3. After pushing the event we could set "value flag" to false. Which makes sense because there is no value (as it was deleted). The user will get exception "value has not been updated" instead of a crash. 

Third option may have some interaction with #241 correction (330a68c6af06eabfe10ce59f0e74a9c03f33e8d2, 799 sourceforge bug), which *removed* unsetting "value flag" in such scenarios. Anyway, this correction looks fishy to me as we can no longer rely on "value flag" to detect if something is stored in the sequence or not.

I'm in favor of option 1. (recommend calling push_event before set_value) + 3. (trigger exception instead of a crash and find another solution for #241).

bourtemb, 2019-12-20:
> Hi @bourtemb

Hi @mliszcz , Thanks for looking into this.
> I've checked the crash from the original post. The problem is that `push_archive_event` postcondition is to delete underlying CORBA sequence: [...]

Indeed!
 
> You can get a similar crash with following read callback:
> 
> ```c++
> void CppTango443::read_value(Tango::Attribute &attr)
> {
>     Tango::DevLong64 x = 0x5555333377771111;
>     *attr_value_read = x;
>     /* attr.set_value(attr_value_read); // commented-out! */
>     attr.set_value_flag(true);
>     attr.set_quality(Tango::ATTR_VALID);
> }
> ```

I understand your point. The state of the attribute object when exiting read_value() is the same at the end as in the code in the issue description.
 
> Now, the solution. I think we can:
> 
>     1. Write in the manual / API docs that any events must be pushed before calling attr.set_value,
> 
>     2. Add new parameter to push_event / new method for pushing event, that will not release the sequence - not tested but should work.
> 
>     3. After pushing the event we could set "value flag" to false. Which makes sense because there is no value (as it was deleted). The user will get exception "value has not been updated" instead of a crash.
> 
> 
> Third option may have some interaction with #241 correction ([330a68c](https://github.com/tango-controls/cppTango/commit/330a68c6af06eabfe10ce59f0e74a9c03f33e8d2), 799 sourceforge bug), which _removed_ unsetting "value flag" in such scenarios. Anyway, this correction looks fishy to me as we can no longer rely on "value flag" to detect if something is stored in the sequence or not.

I agree with you it looks like #241 fix should be improved.

> 
> I'm in favor of option 1. (recommend calling push_event before set_value) + 3. (trigger exception instead of a crash and find another solution for #241).

If we go for this solution, we indeed need another solution for #241 because the problem will reappear. Any idea?
Is there a not too complicated way to detect this kind of situations (the read_attributes case too) so we delete the sequence only when needed? Maybe push_archive_event should not do an attr.set_value and go another way?
Ideally, we would like to be able to invoke push_archive_event from anywhere transparently, without creating memory leak nor causing crashes.
The read_attributes() case described in the issue can be tricky to solve too because the device server programmer does not want to prevent a client to use read_attributes on some attributes and cannot really impose its clients to read the attributes in a certain order...

mliszcz, 2019-12-27:
> If we go for this solution, we indeed need another solution for #241 because the problem will reappear. Any idea?
Is there a not too complicated way to detect this kind of situations (the read_attributes case too) so we delete the sequence only when needed? Maybe push_archive_event should not do an attr.set_value and go another way?

I have not yet investigated what can be done with #241 correction, but, unless we want to break API/contract, we can not change the source of the event data (set with attr.set_value) or stop releasing the sequence after firing an event. `push_archive_event` is just a wrapper for two other public methods. User can call these methods separately, as described in API documentation:
```cpp
{
    Tango::AutoTangoMonitor synch(this);
    att_temp_seq.set_value (temp_seq, 100);
    att_temp_seq.fire_archive_event ();
}
```

So currently the user can expect that `fire_archive_event` reads data from Attribute's sequence and frees the sequence afterwards.

One solution that comes to my mind is to never release the sequence so that the last set value is always available. But I don't know if this may cause any problems and why it was not implemented like that from the very beginning.

bourtemb, 2020-09-10:
> So currently the user can expect that `fire_archive_event` reads data from Attribute's sequence and frees the sequence afterwards.
>
> One solution that comes to my mind is to never release the [...]

I don't know why it was not implemented like that at the beginning.
I think if we could find another way which would allow to have the last attribute set (read and write) value always available, that would be great.
We have regular questions from device servers programmers who are wondering how to get easily access from the device server to the latest attribute read value. There is no easy generic way right now, unless they store this value themselves into a local variable.

As you noted @mliszcz, I think one problem with the current implementation is that the push_archive_event, push_change_event methods where you pass the value to be pushed as parameter are sharing the same buffer.
This is causing some issues, like potential deadlocks in some situation because this imposes to use the Tango Monitor lock to protect the buffer, and leads to some crashes in some situations because the buffer could be deleted by someone else in some situations.

I think one improvement could be to avoid to use that shared buffer when pushing events with values passed as parameter.
This is why I created https://github.com/tango-controls/cppTango/issues/762
But maybe this proposal would also solve some of the limitations which are still present in https://github.com/tango-controls/cppTango/pull/698 PR?

mliszcz, 2020-09-10:
> I think one improvement could be to avoid to use that shared buffer when pushing events with values passed as parameter.

I think we would need a set of new methods for doing that (`push_change_event_safe(value)`), because changing implementation of current `push_change_event` will break existing device servers. Any thoughts on this?

bourtemb, 2020-09-10:
I think there is a problem with the current behaviour of push_change_event methods where you pass the attribute value as parameter.
The user currently does not know (unless by bad experience) that it is using the same buffer as when invoking Attribute::set_value() and it shouldn't have to know.
The expected behaviour for the user is to have a change event sent for this attribute, with the value provided as parameter.
As long as the method does that, I don't expect any complain from the users.
Did I miss something?

mliszcz, 2020-09-10:
I agree with what you said. My previous comment is not correct. What must not be changed is the source of `fire_change_event()` and similar (note fire instead of push in the name), because this one by design is using Attribute's sequence. Currently push_xxx_event methods are simple wrappers around fire_xxx_event.

closed 2022-02-24 by t-b