tango-controls/cppTango #449: BAD_INV_ORDER_ORBHasShutdown when accessing an external proxy from a device server

vxgmichel, 2018-05-03:
It can easily be reproduced using the following pytango server:

```python
import tango.server

proxy = tango.DeviceProxy('sys/database/2')

class Test(tango.server.Device):

    @tango.server.attribute(dtype=str)
    def database_status(self):
        return proxy.status()

if __name__ == '__main__':
    Test.run_server()
```

Exception:

```
DevFailed: DevFailed[
DevError[
    desc = BAD_INV_ORDER CORBA system exception: BAD_INV_ORDER_ORBHasShutdown
  origin = DeviceProxy::status()
  reason = API_CorbaException
severity = ERR]

DevError[
    desc = Failed to execute status() on device (CORBA exception)
  origin = DeviceProxy::status()
  reason = API_CommunicationFailed
severity = ERR]

DevError[
    desc = Failed to read_attribute on device test/nodb/test, attribute database_status
  origin = DeviceProxy::read_attribute()
  reason = API_AttributeFailed
severity = ERR]
]
```

bourtemb, 2018-05-03:
Thanks for the report Vincent.

After talking with @taurel, it seems like your problem is linked to issue #145.
The comments in this ticket explain well what you are seeing.

> It all comes because omniORB manages the ORB object as a singleton. 

To solve #145, the client ORB is destroyed when the device server is started, to get a server ORB.
By doing so, it invalidates all the CORBA objects which were created before the client ORB destruction.

mguijarr, 2018-05-03:
Thanks for the quick help.

If it cannot be fixed, it would be great to display an error message if some *client* objects were alive at the moment of the ORB destruction, so we would be warned.

closed 2022-02-24 by t-b