tango-controls/cppTango #450: 9.3.0 for Debian

picca, 2018-05-08:
Hello guyes.

I would like to know if you will provide later an equivalent of the actual tango sources distribution. Or if we need to start splitting the sources packages for Debian ?

thanks

Frederic

Ingvord, 2018-05-08:
IMHO, it is better to build _good_ Debian packages directly from the repository i.e. cppTango -> libtango,  TangoTest -> tango-test, Database -> tango-db etc and then promote them to the official Debian repositories. Tango source distribution will be preserved to pack everything into a single archive (.zip, .tar.gz) for those who want to build Tango from source.

picca, 2018-05-09:
It is really convenient for me to have only one tarball in order to provide the c++ library and the minimal set of "required" device server to setup a working Tango control system.

Cheers

t-b, 2018-05-09:
@Ingvord I can only second @picca having a single tarball to build from source is very convenient.

Ingvord, 2018-05-11:
@picca , @t-b

I think we  have come to an agreement, as I said: 

> Tango source distribution will be preserved to pack everything into a single archive (.zip, .tar.gz) for those who want to build Tango from source.

:)

t-b, 2019-10-22:
Closing as we have TangoSourceDistribution again.

closed 2019-10-22 by t-b