tango-controls/cppTango #453: cppTango 9.3.0 : Crash of C++ client when trying to subscribe to a Tango 7 device

bourtemb, 2018-05-17:
We experienced a crash of one of our HDB++ Event Subscriber when it was trying to subscribe to archive events on an attribute from a Tango 7 device (which is still using notifd).
The crash was due to one of the assert calls which are at the end of Tango::EventConsumer::initialize_received_from_admin() method.
We could discuss whether we still want to support the compatibility with notifd events but I think attempting to subscribe to some events on Tango 7 device attributes should in any case not stop the clients.
If we decide to stop the support for these old devices (to be discussed), we should at least throw an exception notifying that this is no longer supported without crashing the client.

Ingvord, 2018-05-17:
@bourtemb , my intention was to fail fast. Maybe **assert** is too aggressive in this case. I have replaced **assert** with exception in #454 

In the future IMHO we should remove notifd support from 9LTS

andygotz, 2018-05-18:
@Ingvord before dropping support for notifd events in LTS we need to discuss with the community to see what the impact is and if this is acceptable for the majority ...

Ingvord, 2018-05-18:
@andygotz , sure. We can do it at the next Tango meeting (32)

commit 9535ad1b (2018-05-23): Fix #453: replace assert with exception (#454)

Close #453
closed 2018-05-23 by Ingvord