tango-controls/cppTango #457: cppTango 9.3.0-9.3.1 : Tango::DServer::zmq_event_subscription_change(Tango::DevVarStringArray const*) (eventcmds.cpp:1083)

bourtemb, 2018-05-25:
A memory leak has been introduced in cppTango 9.3.0 and cppTango 9.3.1 (introduced in #423).

The leak has been introduced here:
Tango::DServer::zmq_event_subscription_change(Tango::DevVarStringArray const*) (eventcmds.cpp:1083)

Here is the detail report by valgrind
blocks are definitely lost in loss record 254 of 258
==9156==    at 0x4C298A0: operator new[](unsigned long) (vg_replace_malloc.c:389)
==9156==    by 0x623DF18: CORBA::string_dup(char const*) (in /mntdirect/_segfs/tango/ORB/omniORB4.2.1_p1/debian8/lib/libomniORB4.so.2.1)
==9156==    by 0x55326C2: Tango::DeviceImpl::adm_name() (device.cpp:1663)
==9156==    by 0x55B8306: Tango::DServer::zmq_event_subscription_change(Tango::DevVarStringArray const*) (eventcmds.cpp:1083)
==9156==    by 0x559D6B0: Tango::ZmqEventSubscriptionChangeCmd::execute(Tango::DeviceImpl*, CORBA::Any const&) (dserverclass.cpp:1371)
==9156==    by 0x5579FD3: Tango::DeviceClass::command_handler(Tango::DeviceImpl*, std::string&, CORBA::Any const&) (deviceclass.cpp:1197)
==9156==    by 0x5531ED6: Tango::DeviceImpl::command_inout(char const*, CORBA::Any const&) (device.cpp:1537)
==9156==    by 0x5550665: Tango::Device_2Impl::command_inout_2(char const*, CORBA::Any const&, Tango::DevSource) (device_2.cpp:438)
==9156==    by 0x556B025: Tango::Device_4Impl::command_inout_4(char const*, CORBA::Any const&, Tango::DevSource, Tango::ClntIdent const&) (device_4.cpp:473)
==9156==    by 0x549EDC1: _0RL_lcfn_6fe2f94a21a10053_a3000000(omniCallDescriptor*, omniServant*) (tangoSK.cpp:5383)
==9156==    by 0x6288A50: omniCallHandle::upcall(omniServant*, omniCallDescriptor&) (in /mntdirect/_segfs/tango/ORB/omniORB4.2.1_p1/debian8/lib/libomniORB4.so.2.1)
==9156==    by 0x54A091F: Tango::_impl_Device_4::_dispatch(omniCallHandle&) (tangoSK.cpp:5958)
==9156==    by 0x54A55FD: Tango::_impl_Device_5::_dispatch(omniCallHandle&) (tangoSK.cpp:7478)
==9156==    by 0x628106C: omni::omniOrbPOA::dispatch(omniCallHandle&, omniLocalIdentity*) (in /mntdirect/_segfs/tango/ORB/omniORB4.2.1_p1/debian8/lib/libomniORB4.so.2.1)
==9156==    by 0x625B197: omniLocalIdentity::dispatch(omniCallHandle&) (in /mntdirect/_segfs/tango/ORB/omniORB4.2.1_p1/debian8/lib/libomniORB4.so.2.1)
==9156==    by 0x62A7EEF: omni::GIOP_S::handleRequest() (in /mntdirect/_segfs/tango/ORB/omniORB4.2.1_p1/debian8/lib/libomniORB4.so.2.1)
==9156==    by 0x62A8EF7: omni::GIOP_S::dispatcher() (in /mntdirect/_segfs/tango/ORB/omniORB4.2.1_p1/debian8/lib/libomniORB4.so.2.1)
==9156==    by 0x62A4F24: omni::giopWorker::execute() (in /mntdirect/_segfs/tango/ORB/omniORB4.2.1_p1/debian8/lib/libomniORB4.so.2.1)
==9156==    by 0x624DB05: omniAsyncWorker::real_run() (in /mntdirect/_segfs/tango/ORB/omniORB4.2.1_p1/debian8/lib/libomniORB4.so.2.1)
==9156==    by 0x624EC5E: omniAsyncPoolServer::workerRun(omniAsyncWorker*) (in /mntdirect/_segfs/tango/ORB/omniORB4.2.1_p1/debian8/lib/libomniORB4.so.2.1)
==9156==    by 0x624D74A: omniAsyncWorker::mid_run() (in /mntdirect/_segfs/tango/ORB/omniORB4.2.1_p1/debian8/lib/libomniORB4.so.2.1)
==9156==    by 0x56F6D63: Tango::create_PyPerThData(omni::omniInterceptors::createThread_T::info_T&) (utils.cpp:3255)
==9156==    by 0x624EBF9: omniAsyncWorker::run(void*) (in /mntdirect/_segfs/tango/ORB/omniORB4.2.1_p1/debian8/lib/libomniORB4.so.2.1)
==9156==    by 0x6A5689C: omni_thread_wrapper (in /mntdirect/_segfs/tango/ORB/omniORB4.2.1_p1/debian8/lib/libomnithread.so.4.0)
==9156==    by 0x711E063: start_thread (pthread_create.c:309)
==9156==    by 0x7C3D62C: clone (clone.S:111)

Ingvord, 2018-05-25:
It would be nice to integrate valgrind into CI

bourtemb, 2018-05-30:
Resolved with PR #460

closed 2018-05-30 by bourtemb