tango-controls/cppTango #462: log4tango not build on Windows

liujin001, 2018-06-06:
In ibtango_9.3.1_v120_x64.zip there is not  log4tango.dll/lib

Ingvord, 2018-06-06:
@liujin001 since migration to GitHub log4tango is a part of libtango itself. So the behavior you observer is correct.

closed 2018-06-06 by Ingvord

liujin001, 2018-06-06:
I got it.

When i compile [TangoDatabase](https://github.com/tango-controls/TangoDatabase) (master branch) with libtango_9.3.1_v120_x64, i get link errors, so i first guess the reason is the absence of log4tango library.

`
C:\Users\ics\Dev\TangoDatabase-b\build>cmake .. -G"Visual Studio 12 2013 Win64"
....

C:\Users\ics\Dev\TangoDatabase-b\build>cmake --build ./ --config Release
Microsoft (R) Build Engine version 12.0.31101.0
[Microsoft .NET Framework, version 4.0.30319.42000]
Copyright (C) Microsoft Corporation. All rights reserved.

Build started 2018/6/6 21:36:58.
Project "C:\Users\ics\Dev\TangoDatabase-b\build\ALL_BUILD.vcxproj" on node 1 (default targets).
Project "C:\Users\ics\Dev\TangoDatabase-b\build\ALL_BUILD.vcxproj" (1) is building "C:\Users\ics\Dev\TangoDatabase-b\build\ZERO_CHECK.vcxproj" (2) on node 1 (default targets).
PrepareForBuild:
  Creating directory "x64\Release\ZERO_CHECK\".
  Creating directory "x64\Release\ZERO_CHECK\ZERO_CHECK.tlog\".
InitializeBuildStatus:
  Creating "x64\Release\ZERO_CHECK\ZERO_CHECK.tlog\unsuccessfulbuild" because "AlwaysCreate" was specified.
CustomBuild:
  Checking Build System
  CMake does not need to re-run because C:/Users/ics/Dev/TangoDatabase-b/build/CMakeFiles/generate.stamp is up-to-date.
FinalizeBuildStatus:
  Deleting file "x64\Release\ZERO_CHECK\ZERO_CHECK.tlog\unsuccessfulbuild".
  Touching "x64\Release\ZERO_CHECK\ZERO_CHECK.tlog\ZERO_CHECK.lastbuildstate".
Done Building Project "C:\Users\ics\Dev\TangoDatabase-b\build\ZERO_CHECK.vcxproj" (default targets).
Project "C:\Users\ics\Dev\TangoDatabase-b\build\ALL_BUILD.vcxproj" (1) is building "C:\Users\ics\Dev\TangoDatabase-b\build\Databaseds.vcxproj" (3) on node 1 (default targets).
PrepareForBuild:
  Creating directory "Databaseds.dir\Release\".
  Creating directory "C:\Users\ics\Dev\TangoDatabase-b\build\Release\".
  Creating directory "Databaseds.dir\Release\Databaseds.tlog\".
InitializeBuildStatus:
  Creating "Databaseds.dir\Release\Databaseds.tlog\unsuccessfulbuild" because "AlwaysCreate" was specified.
CustomBuild:
  Building Custom Rule C:/Users/ics/Dev/TangoDatabase-b/CMakeLists.txt
  CMake does not need to re-run because C:/Users/ics/Dev/TangoDatabase-b/build/CMakeFiles/generate.stamp is up-to-date.
ClCompile:
  C:\Program Files (x86)\Microsoft Visual Studio 12.0\VC\bin\x86_amd64\CL.exe /c /I"C:\Users\ics\Dev\TangoDatabase-b\." /I"C:\mysql-5.7.22-winx64\include" /IC:\libtango_9.3.1_v120_x64\include /nologo /W3 /WX- /O2 /Ob2 /D WIN32 /D _WINDOWS /D NDEBUG /D "CMAKE_INTDIR=\"Release\"" /D _MBCS /Gm- /EHsc /MD /GS /fp:precise /Zc:wchar_t /Zc:forScope /GR /Fo"Databaseds.dir\Release\\" /Fd"Databaseds.dir\Release\vc120.pdb" /Gd /TP /errorReport:queue "C:\Users\ics\Dev\TangoDatabase-b\DataBase.cpp" "C:\Users\ics\Dev\TangoDatabase-b\DataBaseClass.cpp" "C:\Users\ics\Dev\TangoDatabase-b\DataBaseStateMachine.cpp" "C:\Users\ics\Dev\TangoDatabase-b\ClassFactory.cpp" "C:\Users\ics\Dev\TangoDatabase-b\main.cpp" "C:\Users\ics\Dev\TangoDatabase-b\DataBaseUtils.cpp" "C:\Users\ics\Dev\TangoDatabase-b\update_starter.cpp"
  DataBase.cpp
  DataBaseClass.cpp
  DataBaseStateMachine.cpp
  ClassFactory.cpp
  main.cpp
  DataBaseUtils.cpp
  update_starter.cpp
  Generating Code...
Link:
  C:\Program Files (x86)\Microsoft Visual Studio 12.0\VC\bin\x86_amd64\link.exe /ERRORREPORT:QUEUE /OUT:"C:\Users\ics\Dev\TangoDatabase-b\build\Release\Databaseds.exe" /INCREMENTAL:NO /NOLOGO /LIBPATH:"C:/mysql-5.7.22-winx64/lib" /LIBPATH:"C:/mysql-5.7.22-winx64/lib/Release" /LIBPATH:C:/libtango_9.3.1_v120_x64/lib /LIBPATH:C:/libtango_9.3.1_v120_x64/lib/Release mysqlclient.lib tango.lib omniORB4_rt.lib omniDynamic4_rt.lib COS4_rt.lib omnithread_rt.lib "libzmq-v120-mt-4_0_5.lib" kernel32.lib user32.lib gdi32.lib winspool.lib shell32.lib ole32.lib oleaut32.lib uuid.lib comdlg32.lib advapi32.lib /MANIFEST /MANIFESTUAC:"level='asInvoker' uiAccess='false'" /manifest:embed /PDB:"C:/Users/ics/Dev/TangoDatabase-b/build/Release/Databaseds.pdb" /SUBSYSTEM:CONSOLE /TLBID:1 /DYNAMICBASE /NXCOMPAT /IMPLIB:"C:/Users/ics/Dev/TangoDatabase-b/build/Release/Databaseds.lib" /MACHINE:X64  /machine:x64 Databaseds.dir\Release\DataBase.obj
  Databaseds.dir\Release\DataBaseClass.obj
  Databaseds.dir\Release\DataBaseStateMachine.obj
  Databaseds.dir\Release\ClassFactory.obj
  Databaseds.dir\Release\main.obj
  Databaseds.dir\Release\DataBaseUtils.obj
  Databaseds.dir\Release\update_starter.obj
     Creating library C:/Users/ics/Dev/TangoDatabase-b/build/Release/Databaseds.lib and object C:/Users/ics/Dev/TangoDatabase-b/build/Release/Databaseds.exp
DataBaseClass.obj : error LNK2019: unresolved external symbol "class log4tango::Logger * Tango::_core_logger" (?_core_logger@Tango@@3PEAVLogger@log4tango@@EA) referenced in function "protected: __cdecl DataBase_ns::DataBaseClass::DataBaseClass(class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > &)" (??0DataBaseClass@DataBase_ns@@IEAA@AEAV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@@Z) [C:\Users\ics\Dev\TangoDatabase-b\build\Databaseds.vcxproj]
main.obj : error LNK2001: unresolved external symbol "class log4tango::Logger * Tango::_core_logger" (?_core_logger@Tango@@3PEAVLogger@log4tango@@EA) [C:\Users\ics\Dev\TangoDatabase-b\build\Databaseds.vcxproj]
update_starter.obj : error LNK2001: unresolved external symbol "class log4tango::Logger * Tango::_core_logger" (?_core_logger@Tango@@3PEAVLogger@log4tango@@EA) [C:\Users\ics\Dev\TangoDatabase-b\build\Databaseds.vcx [... cut ...]
  main.obj : error LNK2019: unresolved external symbol "public: static char const * const Tango::Device::_PD_repoId" (?_PD_repoId@Device@Tango@@2PEBDEB) referenced in function "public: class Tango::_objref_Device * __cdecl POA_Tango::Device::_this(void)" (?_this@Device@POA_Tango@@QEAAPEAV_objref_Device@Tango@@XZ) [C:\Users\ics\Dev\TangoDatabase-b\build\Databaseds.vcxproj]
  C:\Users\ics\Dev\TangoDatabase-b\build\Release\Databaseds.exe : fatal error LNK1120: 3 unresolved externals [C:\Users\ics\Dev\TangoDatabase-b\build\Databaseds.vcxproj]

    0 Warning(s)
    7 Error(s)

Time Elapsed 00:00:24.93
`

Ingvord, 2018-06-06:
OK. I would move this to TangoDatabase then.

@NexeyaSGara, maybe you could comment on this?

NexeyaSGara, 2018-06-07:
Hi, 

I don't know if the CMakelist that you are using is compatible with Windows.

Can you check inside the generated vcxproj if you have reference to the librairies ?

liujin001, 2018-06-08:
Hi,
I’ll check vcxproj, I add the CMakeLists.txt file used in my test(based on TangoDatabase/CMakeLists.txt), this CMakeLists file can successfully make TangoDatabase with tango-9.2.2, but link error with libtango_9.3.1_v120_x64

```cmake
cmake_minimum_required(VERSION 2.8.9)
project(DataBase)
 
#include(configure/CMakeLists.txt)
 
set(MAJOR_VERSION "5")
set(MINOR_VERSION "6")
set(PATCH_VERSION "0")
set(PROJECT_VERSION "${MAJOR_VERSION}.${MINOR_VERSION}.${PATCH_VERSION}")
 
set(SOURCES DataBase.cpp
            DataBaseClass.cpp
            DataBaseStateMachine.cpp
            ClassFactory.cpp
            main.cpp)
 
set(ADDITIONAL_SOURCES  DataBaseUtils.cpp
                        update_starter.cpp)
 
# use libtango_9.3.1_v120_x64.zip from GitHub cppTango CI, fail for link error!!
set(INCLUDEPATH "C:/mysql-5.7.22-winx64/include" "C:/libtango_9.3.1_v120_x64/include")
set(LIBPATH "C:/mysql-5.7.22-winx64/lib" "C:/libtango_9.3.1_v120_x64/lib")
set(EXTRALIB mysqlclient tango omniORB4_rt omniDynamic4_rt COS4_rt omnithread_rt libzmq-v120-mt-4_0_5)
# Use tango-9.2.2 install file from sourceforge,  success!!
#set(INCLUDEPATH "C:/mysql-5.7.22-winx64/include" "C:/tango-9.2.2/win64/include/vc12")
#set(LIBPATH "C:/mysql-5.7.22-winx64/lib" "C:/tango-9.2.2/win64/lib/vc12_dll")
#set(EXTRALIB mysqlclient tango omniORB4_rt omniDynamic4_rt COS4_rt omnithread_rt zmq log4tango)
 
#include_directories("." ${TANGO_PKG_INCLUDE_DIRS} ${MYSQL_INCLUDE_DIRS})
#link_directories(${TANGO_PKG_LIBRARY_DIRS})
include_directories("." ${INCLUDEPATH})
link_directories(${LIBPATH})
 
add_executable(Databaseds ${SOURCES} ${ADDITIONAL_SOURCES})
#target_link_libraries(Databaseds ${TANGO_PKG_LIBRARIES} ${MYSQL_LIBRARIES} -Wl,-z,now -pie)
target_link_libraries(Databaseds ${EXTRALIB})
#target_compile_options(Databaseds PUBLIC ${TANGO_PKG_CFLAGS_OTHER} -Wall -Wextra -D_FORTIFY_SOURCE=2 -O1 -fpie)
 
 
#install(TARGETS Databaseds
#        RUNTIME DESTINATION "${CMAKE_INSTALL_FULL_BINDIR}"
#        CONFIGURATIONS ${CMAKE_BUILD_TYPE})
```