tango-controls/cppTango #463: Changing a polling period of a dynamic attribute breaks device restart functionality

dvucko, 2018-07-03:
It appears there is a bug in Tango (v9.2.5) related to dynamic attributes. It can be reproduced as follows:

1. Use Pogo to create a new device server with a polled dynamic scalar attribute.
2. Create a sample dynamic attribute in code: 
```
void TangoTestDevice::add_dynamic_attributes()
{
	/*----- PROTECTED REGION ID(TangoSandbox::add_dynamic_attributes) ENABLED START -----*/
		
	add_DynamicTestAttribute_dynamic_attribute("Dynamic Attribute 1");
	
	/*----- PROTECTED REGION END -----*/	//	TangoSandbox::add_dynamic_attributes
}
```
3. Start the device server. Use Jive to change the polling period to another value.
4. Try to restart the device server by right-clicking on the instance in Jive and selecting _Restart device_. Doing so will generate a message box with the following Tango error:
```
Desc -> Device a/b/c is not polled
Reason -> API_DeviceNotPolled
Origin -> DServer::upd_obj_polling_period
Desc -> init_deivce() method for device a/b/c throws DevFailed exception. Device not available any more. Please fix exception reason
Reason -> API_InitThrowsException
Origin -> Dserver::restart()
Desc -> Cannot execute command DevRestart on dserver/name/a
Reason -> TangoApi_CANNOT_EXECUTE_COMMAND
Origin -> Connection.command_inout()
```
My investigation shows that the culprit is the `Tango::Attr.set_polling_period()` method which, as soon as it is used, causes restart to no longer work. I assume that changing the polling period via Jive  calls `set_polling_period()` in the background and thus breaks the restart.

Even after quitting and re-starting the device server, _Restart device_ will not work unless the instance is deleted and recreated.

This may be related to another dynamic attributes issue I mentioned in the forums:
[Polling properties for dynamically created attributes are ignored](http://www.tango-controls.org/community/forum/c/development/c/polling-properties-for-dynamically-created-attributes-are-ignored/)

Is there any workaround for this?

bourtemb, 2018-07-05:
This is indeed related to the other dynamic attribute issue #427.
Applying the patch from issue #427 should be enough if you can recompile the tango library.
I tested it on a simple example and it seemed to work.

bourtemb, 2018-07-05:
To be accurate, here is the patch (with line numbers corresponding to tango 9.2.5:

```
--- server/device.cpp.orig	2018-07-05 21:45:51.809102000 +0200
+++ server/device.cpp	2018-07-05 21:50:24.372749000 +0200
@@ -3245,8 +3245,7 @@
 //
 
     long per = new_attr->get_polling_period();
-    Tango::Util *tg = Tango::Util::instance();
-    if (tg->is_svr_starting() == false && per != 0)
+    if ((!is_attribute_polled(attr_name)) && (per != 0))
         poll_attribute(attr_name,per);
 
 //

```

bourtemb, 2019-03-01:
Should be fixed with #427

closed 2019-03-01 by bourtemb