tango-controls/cppTango #466: Events skipped in specific cases (after device server restart)

mkomel, 2018-07-30:
We have encountered a problem with the Tango eventing system (Tango v9.2.5, Ubuntu 18.04 LTS).

In summary:
if a client is subscribed to attributes on a device server and this device server is stopped and soon afterwards restarted, it can happen that some events are skipped (not sent from the device server at all).

I have prepared a minimal example to reproduce the issue:
- there is a simple device server with two polled attributes (1000ms): state and SimpleScalar.
  This device server spends first two polling loops for setting `state=DevState.UNKNOWN`; `SimpleScalar=0` and afterwards `=1`
  After that it will just continue setting `state=DevState.ON; SimpleScalar=999`

- a client subscribes to these two attributes (in stateless mode) and stores the values received
- the device server is started, and a few seconds afterwards the events start coming to the client
- after 15 seconds (this should be plenty of time for the eventing system to become fully operational) the device server is stopped...
- client checks whether the last received values (via events) match with the current values (via direct read_attributes call) on the device server;  if they do not match, wait for another 15 seconds and check again; if they still do not match, it is an issue reproduced
- after 4-8 seconds the device server is started again and the procedure repeated

Depending on the timing between stopping and restarting the device server (it seems so), it can happen that the client will never receive the last values from the server (`DevState.ON; 999`), but will be "stuck" with an outdated values (`DevState.UNKNOWN; 1 or 2`).

I'm attaching here a zip file ([tangoeventingissues.zip](https://github.com/tango-controls/cppTango/files/2240993/tangoeventingissues.zip)) with:
- import_testds.py:   execute this to import the test data server to the Tango database
- testscript.py:   execute this to start the testing loop
- deviceserver/*:  device server (be sure to build it first before starting the testing loop script)

First execute _import_testds.py_ to import the device server to the Tango database.
Afterwards start _testscript.py_. Do not start the device server, the testing script starts and stops the device server by its own inside the testing loop.
The testing loop can take up to 10 minutes to hit the issue. Usually it took ~1-5minutes on my machine.

Please let me know if you can reproduce this issue with the provided example. 
If not, I guess it could be related to the exact environment (for example ZMQ version). 
If yes, it might be a bug in Tango.

mkomel, 2018-07-30:
I just noticed there already was an issue report about this problem some time ago: #359

Is there a known workaround or has anything already been done about this (so that the next Tango release contains a fix for this)?

bourtemb, 2018-08-06:
Hi Matej,

Thank you very much for the detailed bug report.
This issue indeed seems to be the same as #359.
There is no workaround without recompiling cppTango and applying the patch I suggested in the last comment I made in #359.
We agreed during a Tango kernel teleconf to implement the fix I suggested at that time, even though this would trigger some events in some cases even if the value did not change. It is more important to ensure that the event with the latest value is correctly sent.
As far as I know, it has not been done yet but the fix is known so the fix will be present in a future release of Tango because this bug can be critical in some situations.

I close this issue as duplicate of #359.

closed 2018-08-06 by bourtemb