tango-controls/cppTango #471: Setting R/W array atttribute write value with set_write_value() not working

mike-a-smith, 2018-08-14:
Calling set_write_value() for a WAttribute does not work for DevVarDoubleArray type.  For scalar types it seems to work OK.  The proper ptr variable of WAttribute is set within the method and data is copied; however, on exit of the set_write_value() method the memory it points to is no longer valid.  Seems to be a resurfacing of this old bug:  https://sourceforge.net/p/tango-cs/bugs/53/

When I try the suggested work-around for this old bug in my code, the values are copied properly.

		// load a default filter on first contact (filter is changeable via coefficients attr)
		std::vector<double> defCoeffs(nFFT * nTAPS, 1.0);// = gen_pfb_coeff(nFFT, nTAPS);

		Tango::WAttribute &coeff_attr = get_device_attr()->get_w_attr_by_name("coefficients");
		coeff_attr.set_write_value(defCoeffs.data(), nFFT * nTAPS);
		// double_ptr now points to some garbage data
		// work-around for set_write_value() bug https://sourceforge.net/p/tango-cs/bugs/53/
		CORBA::Any an;
		Tango::DevVarDoubleArray tmpbuf(nFFT*nTAPS, nFFT*nTAPS, defCoeffs.data());
		an <<= tmpbuf;
		coeff_attr.copy_data(an);
		// double_ptr points to valid data
		// load
		INFO_STREAM << "Loading default PFB coefficients" << endl;
		write_coefficients(coeff_attr);

bourtemb, 2018-09-13:
Hi,

I think there is a confusion here because of the name of the methods...
When you write 

> on exit of the set_write_value() method the memory it points to is no longer valid

 and then you put in comment in your code sample:
`// double_ptr now points to some garbage data`

It looks like you tried to use 
`void get_write_value(const Tango::DevDouble *&ptr)`
method to retrieve the set value of your attribute.

From what I understood, this get_write_value() method is supposed to be used only when this attribute is written by a remote client. It is used in _YourTangoClass::write_YourDoubleArrayAttribute()_ method to retrieve the set value set by the client which executed _write_attribute("YourDoubleArrayAttribute", ...)_

You should not use get_write_value outside this _YourTangoClass::write_YourDoubleArrayAttribute()_ method.

If you do:

```
std::vector<double> defCoeffs(nFFT * nTAPS, 1.0);
Tango::WAttribute &coeff_attr = get_device_attr()->get_w_attr_by_name("coefficients");
coeff_attr.set_write_value(defCoeffs.data(), nFFT * nTAPS);
```

and read the _coefficients_ attribute after that with tg_devtest (Test device panel from jive), you should see the set value for _coefficients_ attribute is actually correct (at least this is what I see).
_WAttribute::double_ptr_ is not the _WAttribute_ member which is used to store the attribute set value, _double_ptr_ is used to store a new set value coming from a _write_attribute_ request.
The set value seems to be stored in _WAttribute::double_array_val_ instead.

Could you please check that when you read your attribute after having executed set_write_value, you get the correct values in the set point part of the result?

Hoping this helps,
Reynald

bourtemb, 2019-03-01:
@mike-a-smith , did you try my suggestions?
Do you need some help?
Any news on this issue?

mike-a-smith, 2019-03-01:
@bourtemb, thanks for your explanation.  Unfortunately, it's been on my todo list and I haven't tried it yet.  I'll try to find some time to test this out so you can close out this issue.

mike-a-smith, 2019-03-08:
@bourtemb I tried your suggestions and I do get the correct set point values when I read the attribute from an external client.

bourtemb, 2019-03-11:
@mike-a-smith, thanks for your feedback. So set_write_value() is actually working as expected when setting a R/W array. There was just a confusion with the role of the get_write_value() method it seems.
Can we close this issue? Do you need more help on this topic?

mike-a-smith, 2019-03-11:
@bourtemb Yes, please close the issue.

closed 2019-03-11 by mike-a-smith