tango-controls/cppTango #487: Tango 9.3.2 - tests do not compile with old compilers (at least gcc <= 4.4.5)

bourtemb, 2018-09-17:
PR #476 (Backport #275: split Runner) introduced some C++11 code without adding alternative code compatible with old compilers (yes, some institutes are still using these old compilers).
As a result,
```
mkdir build; cd build
cmake ..
make install
```
fails with compilation errors.

A work around to compile and install the tango 9.3.2 library and includes on hosts using these old compilers is to simply comment line 36 in the top CMakeLists.txt to avoid the compilation of the tests:
https://github.com/tango-controls/cppTango/blob/9.3.2/CMakeLists.txt#L36

t-b, 2019-06-05:
Closing as #275 was not merged and we don't care about master for now.

closed 2019-06-05 by t-b

bourtemb, 2019-06-07:
> Closing as #275 was not merged and we don't care about master for now.

#275 has been _backported_ into tango-9-lts via #476 (which was merged!) so this issue still exists with cppTango 9.3.3 on old compilers and is still present in cppTango 9.3.3.
We can mark it as won't fix though because we don't want to spend some time to fix tests for these old compilers.
Please re-open if you consider this as critical.