tango-controls/cppTango #491: Memory leak in cppTango 8.1.2

DmitryKuk, 2018-10-11:
Hello, everyone,

I am trying to debug memory leaks in my TDS running on two different linux systems with different libtango8-dev packages:
1. x86_64 *buntu:
```
$ apt-cache show libtango8-dev 
Package: libtango8-dev
Status: install ok installed
Priority: extra
Section: libdevel
Installed-Size: 1537
Maintainer: Ubuntu Developers <ubuntu-devel-discuss@lists.ubuntu.com>
Architecture: amd64
Multi-Arch: same
Source: tango
Version: 8.1.2c+dfsg-7ubuntu3
Depends: libcos4-dev, liblog4tango5-dev, libtango8v5 (= 8.1.2c+dfsg-7ubuntu3), libzmq5-dev
Conflicts: libtango7-dev
Description: TANGO distributed control system - development library
 TANGO is an object-oriented distributed control system using CORBA. In
 TANGO all objects are representations of devices, which can be on the
 same computer or distributed over a network. Communication between
 devices uses CORBA and can be synchronous, asynchronous or event driven.
 .
 The object model in TANGO supports methods, attributes and properties.
 TANGO provides an API which hides all the details of network access and
 provides object browsing, discovery, and security features. Permanent
 data is stored in a MySQL database.
 .
 TANGO is being actively developed as a collaborative effort between the
 ESRF (www.esrf.eu), Soleil (synchrotron-soleil.fr), Alba (www.cells.es),
 and Elettra institutes (www.elettra.trieste.it).
 .
 This package contains the tango8 development files.
Description-md5: 232fc1dc6796ef640371cafdfada8b40
Homepage: http://www.tango-controls.org
Original-Maintainer: Debian Science Maintainers <debian-science-maintainers@lists.alioth.debian.org>
```
(looks like this package built with some patches applied, including p812_{1,2,3,4}.diff: https://launchpad.net/ubuntu/+source/tango/8.1.2c+dfsg-7ubuntu3)

2. Some Debian8 on MIPS:
```
$ apt-cache show libtango8-dev
Package: libtango8-dev
Source: tango
Version: 8.1.2c+dfsg-5
Installed-Size: 1537
Maintainer: Debian Science Maintainers <debian-science-maintainers@lists.alioth.debian.org>
Architecture: mipsel
Depends: libcos4-dev, liblog4tango5-dev, libtango8 (= 8.1.2c+dfsg-5), libzmq3-dev
Conflicts: libtango7-dev
Description-en: TANGO distributed control system - development library
 TANGO is an object-oriented distributed control system using CORBA. In
 TANGO all objects are representations of devices, which can be on the
 same computer or distributed over a network. Communication between
 devices uses CORBA and can be synchronous, asynchronous or event driven.
 .
 The object model in TANGO supports methods, attributes and properties.
 TANGO provides an API which hides all the details of network access and
 provides object browsing, discovery, and security features. Permanent
 data is stored in a MySQL database.
 .
 TANGO is being actively developed as a collaborative effort between the
 ESRF (www.esrf.eu), Soleil (synchrotron-soleil.fr), Alba (www.cells.es),
 and Elettra institutes (www.elettra.trieste.it).
 .
 This package contains the tango8 development files.
Description-md5: 232fc1dc6796ef640371cafdfada8b40
Multi-Arch: same
Homepage: http://www.tango-controls.org
Tag: devel::library, role::devel-lib
Section: libdevel
Priority: extra
Filename: pool/main/t/tango/libtango8-dev_8.1.2c+dfsg-5_mipsel.deb
Size: 144166
MD5sum: 1943b0f28fc79e00d265743011d05b85
SHA1: 81c58e54208579f3a32e8ff7967c03f0c5cf8110
SHA256: 038b17c0295cd52bbe4ab23b8a94c6fdcfeeb8948e1d7d09c7a8359a452de4d3
```

On both systems I see process' resident and virtual memory grows (faster or slower).
After some tests with valgrind's massif tool and different device servers (including TangoTest and clean TDS, generated by POGO), I got same bad result (see images)...

Used TangoTest from https://github.com/tango-controls/TangoTest, TangoTest-Release-1.9.0.
[image]
[image]
[image]

Is it well-known bug of Tango 8? Have anybody patches for it?

Thanks!

DmitryKuk, 2018-10-11:
P.S. Valgrind also says this is not "definitely loss", all memory correctly freed at process exit.

bourtemb, 2018-10-22:
Hi,

I suppose your 3 images are showing the memory consumption for 3 different device servers (Tangotest 1.9.0 and 2 of your device servers). What is the unit of the time (it is written "time in i" which is quite confusing)?
The 3 images are quite different so it would be interesting to understand the differences between these 3 device servers and on the clients which are connected to them. This could help to better identify where it is "leaking" memory. What happened in the second case where a lot of memory got freed? Was the device server simply killed?
It looks like most of the memory is used by zmq in these 3 cases.
It would be interesting to know the differences on the client side too.
How many clients do you have in these 3 different cases? Are you using generic clients like atkpanel or some specifically developed client applications (in which language are they written)? At what speed do you generate events, on which attributes?
Did you try with the latest release of TangoTest, and the latest release of the tango library (latest official release is 9.2.5)?
Some bugs have been fixed between Tango 8 and Tango 9.2.5 so I can only encourage you to upgrade your Tango version and TangoTest version or to have a look at the commit logs to see what has been fixed since Tango 8 (if the problem is fixed).

bourtemb, 2019-03-01:
@DmitryKuk, any news on this issue?

t-b, 2019-06-05:
No answer for three months. Closing for now. Feel free to reopen any time.

closed 2019-06-05 by t-b