tango-controls/cppTango #501: Event subscriber exit after device server restart on Windows

liujin001, 2018-11-24:
I test libtango_9.3.3_v140_x64.zip (download from https://ci.appveyor.com/project/bourtemb/cpptango-tt3w8/branch/tango-9-lts ) on Windows 10 and Visual Studio Community 2015.

1. start TangoTest/test server, config sys/tg_test/1 device's long_scalar attribute Periodic event
2. start event subscriber(EventSub.exe  
[EventSub.cpp.zip](https://github.com/tango-controls/cppTango/files/2612375/EventSub.cpp.zip)
)
3. shut down TangoTest/test server(Ctrl+c), and re-run TangoTest/test server instantly
4. EventSub.exe report API_EventTimeout exception one or two times, then exit

but i test this on CentOS7(tango-9.2.5a),  event subscriber will not exit, after TangoTest/test restart, subscriber receive event normally.

bourtemb, 2018-11-27:
Hi @liujin001 

I'm sorry but I didn't manage to reproduce the error you encountered. In my environment, EventSub.exe does not exit when I restart TangoTest and reconnects well.
Is it systematically reproducible in your case?
If yes, could you give more details about your configuration?
Are TangoTest.exe and EventSub.exe running on the same host/same OS in your case?
Are they both using libtango_9.3.3_v140_x64 version?
What is your long_scalar periodic event period? What is long_scalar attribute polling period?
Sorry to ask this question, but are you sure the EventSubscriber didn't exit on Windows simply because the 10 minutes of the tests were expired?
Can you check the return code of EventSub.exe to see whether it is EXIT_FAILURE or EXIT_SUCCESS?
Could you provide the log of the messages displayed on the console?
Thanks in advance for your answers.

liujin001, 2018-11-30:
Hi bourtemb

It's maybe my Win10 Virtual Machine problem,  because if i run TangoTest.exe and EventSub.exe on Win10 Physical Machine, it not reproduce this problem.

My VM test environment information:
1. build TangoTest.exe and EventSub.exe all with libtango_9.3.3_v140_x64
2. run TangoTest.exe and EventSub.exe on one Win10 Virtual Machine.
3. long_scalar periodic event period is 1000ms
4. i am sure EventSub.exe not exit on Windows simply because the 10 minutes of the tests were expired

I will check my VM config, then may give more information.

bourtemb, 2019-03-01:
@liujin001, any update on this issue?

t-b, 2020-10-20:
Closing as it the OP could not reproduce the problem. Please reopen if you have any news.

closed 2020-10-20 by t-b