tango-controls/cppTango #506: Exception in init_device() blocks server

tiagocoutinho, 2018-12-14:
Throwing an exception in my device server in the device `init_device()`:
```c++
void TangoTest::init_device() {
    throw "error!";
}
```
blocks if I my `main()` looks like this:

```c++
int main(int argc, char *argv[]) {
    Tango::Util *util = NULL;
    try {
        util = Tango::Util::init(argc,argv);
        util->server_init(false);
        cout << "Ready to accept request" << endl;
        util->server_run();
    } catch (...) {
        cout << "Received an unknown exception" << endl;
	cout << "Exiting" << endl;
    }
    util->server_cleanup();
    delete util;
    return(0);
}
```
The difference from a pogo generated server is that I properly call the `delete util`.
I notice the server is blocked in the `Util::~Util()` destructor. It looks like it is waiting for the Polling thread. Seems like a dead lock to me.

In cases where the server stops *orderly* there seems to be no problem in calling `Util::~Util()`.

I report this because in PyTango I notice the same behaviour. In PyTango, the python Util object calls the underlying C++ Util destructor *automatically* so if the `init_device()` throws an exception, PyTango servers always block.

bourtemb, 2018-12-17:
Hi @tiagocoutinho,

Thanks for the bug report.
I didn't manage to reproduce your issue on Linux...
Are you on Windows?
What version of cppTango are you using?

tiagocoutinho, 2018-12-17:
Hi @bourtemb,

I am using linux mint 19.
I tested with Tango C++ 9.2.5 and 9.3.2alpha and both show the problem.

bourtemb, 2018-12-17:
OK, thanks!
I managed to reproduce the problem with a Linux Mint 19 - 64 bits VM and libtango 9.2.5 installed from the Debian package.
So, on Linux Mint 19, we have:
- libzmq 4.2.5
- omniORB 4.2.2
- g++ 7.3.0
To be continued...

closed 2022-02-24 by t-b