tango-controls/cppTango #507: events and TangoAccessControl

egorbe, 2018-12-20:
Hello,
We use TangoAccessControl and I had been experiencing problems with events for a long time. C++/Python clients could sucessfully subscribe to events (i used periodic events) but they stopped arriving after 600 seconds.  
It turned out that clients should resubscribe every 200 seconds by executing EventConfirmSubscription command on the corresponding administrative device. However, the command was not in the list of allowed commands for class Dserver so it was silently rejected by TangoAccessControl.
After I added the command to the list of allowed commands, everything starts to work just fine.

```
MySQL [tango]> select * from property_class where value='EventConfirmSubscription';
+---------+------------------+-------+--------------------------+---------------------+---------------------+---------+
| class   | name             | count | value                    | updated             | accessed            | comment |
+---------+------------------+-------+--------------------------+---------------------+---------------------+---------+
| DServer | AllowedAccessCmd |     2 | EventConfirmSubscription | 2018-12-19 10:38:45 | 2018-12-19 10:38:45 | NULL    |
+---------+------------------+-------+--------------------------+---------------------+---------------------+---------+

```
I think it would be nice if try/catch block in client/eventkeepalive.cpp would report about denied access instead of silent ommision. Also, I guess the dserver EventConfirmSubscription command should be allowed by default in TANGO database by create/update database scripts.

thanks,
Evgeny

bourtemb, 2019-01-04:
Hi Evgeny,

Thank you very much for the detailed report.
Indeed Tango C++ device servers expect the clients to confirm their subscription at least once every 10 minutes. As you noticed, the C++ clients will do so by executing the EventConfirmSubscription command on the admin device every 200 seconds (10min / 3).

This made us realize that our setup is actually not correct in our institute!
So I thought it is very surprising that nobody ever complained to us about it...
After deeper investigations, I realized that this EventConfirmSubscription command is not used by the Java clients which seem to prefer the ZmqEventSubscriptionChange command so the problem is not present in Java Tango clients, which represents most of our clients in our institute. This is probably the reason why nobody ever complained in our institute. The C++ clients being most of the time device servers in our case, they have special TAC permissions or they are simply not using events in our case.

> I think it would be nice if try/catch block in client/eventkeepalive.cpp would report about denied access instead of silent ommision.

I agree. We could trigger an error event in this case indeed.

> Also, I guess the dserver EventConfirmSubscription command should be allowed by default in TANGO database by create/update database scripts.

I think it would make sense indeed, unless you really want the non-authorized people to be able to receive events only during the first 10 minutes, which is probably not the majority of the use cases, because as you noticed, this can be very confusing! So I agree with you.

bourtemb, 2019-01-09:
I created https://github.com/tango-controls/TangoDatabase/pull/13 to allow dserver EventConfirmSubscription command by default in TangoDatabase database creation/update scripts.

The generation of an error event still needs to be done.

closed 2022-02-24 by t-b