tango-controls/cppTango #509: Tango9.3.2 cpp read_attribute no more DevFail on ATTR_INVALID

alexElettra, 2019-01-07:
Same server e same client compiled for Tango9,
a client reading its attribute being ATTR_INVALID:

1) when using tango-9.2.5a/lib/libtango.so.9 the client is notified with a DevFail:
       Tango exception
        Severity = ERROR 
        Error reason = API_EmptyDeviceAttribute
        Desc : cannot extract, no data in DeviceAttribute object 
        Origin : DeviceAttribute::is_empty

**2) when using tango-9.3.2/lib/libtango.so.9 the client IS NO MORE notified with DevFail:
     --------------------------------
     No data in DeviceAttribute object
     --------------------------------**

Is this the new behavior? or a bug? or I missed the relevant line in the docs?

Thanks,
Alessandro

bourtemb, 2019-01-07:
Hi @alexElettra 

I think this is a side effect of https://github.com/tango-controls/cppTango/pull/393 which was done to solve https://github.com/tango-controls/cppTango/issues/392.
Were you getting this exception when invoking DeviceAttribute::get_type() method?

alexElettra, 2019-01-07:
Hi @bourtemb 

a very/very short snippet of my code:

try {
    Tango::DeviceAttribute attr = deviceproxy->read_attribute(...);
    attr_type=attr.get_type();
    ....
    switch (attr_type){
      .... since attribute may be of different type, do proper way....
    }
} catch (Tango::DevFailed &e) {
 .... here I fall if using Tango 9.2.5a ....
} catch(...) {
}

..... while in Tango 9.3.2 read_attribute of a ATTR_INVALID do not send a DevFail.

So I see now that wasn't the read_attribute() but attr.get_type() leading me to DevFail!
In the docs: read_attribute(): throws ConnectionFailed, CommunicationFailed... never DevFail! so I was driven into a mistake by the change in get_type()!

Perhaps this change may be underlined in RELEASE_NOTES doc.

bourtemb, 2019-01-07:
> Perhaps this change may be underlined in RELEASE_NOTES doc.

I agree.

bourtemb, 2019-03-29:
This DeviceAttribute::get_type() behaviour change is now mentioned in the changelog and release notes and the potential consequences are now listed there.
Solved in #519

closed 2019-03-29 by bourtemb