tango-controls/cppTango #511: Possible crash due to race condition when pushing events

gscalamera, 2019-01-10:
If using push_change_event/push_archive_event in the code with also the poller enabled and event thresholds set, the device server could crash.
Attached a simple device server reproducing the issue.

Some backtrace of the crash follows:

#0  tcache_get (tc_idx=0) at malloc.c:2943
#1  __GI___libc_malloc (bytes=16) at malloc.c:3050
#2  0x00007fa6c857f1a8 in operator new(unsigned long) () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#3  0x00007fa6c92d8aea in __gnu_cxx::new_allocator<double>::allocate (this=0x7fa6c235ab00, __n=2) at /usr/include/c++/7/ext/new_allocator.h:111
#4  0x00007fa6c92d7ef8 in std::allocator_traits<std::allocator<double> >::allocate (__a=..., __n=2) at /usr/include/c++/7/bits/alloc_traits.h:436
#5  0x00007fa6c92d6fda in std::_Vector_base<double, std::allocator<double> >::_M_allocate (this=0x7fa6c235ab00, __n=2) at /usr/include/c++/7/bits/stl_vector.h:172
#6  0x00007fa6c9503764 in std::vector<double, std::allocator<double> >::_M_realloc_insert<double> (this=0x7fa6c235ab00, __position=1.4440053121040064e+214, __args#0=@0x7fa6c235aa50: 0)
    at /usr/include/c++/7/bits/vector.tcc:406
#7  0x00007fa6c9502fb0 in std::vector<double, std::allocator<double> >::emplace_back<double> (this=0x7fa6c235ab00, __args#0=@0x7fa6c235aa50: 0) at /usr/include/c++/7/bits/vector.tcc:105
#8  0x00007fa6c9502746 in std::vector<double, std::allocator<double> >::push_back (this=0x7fa6c235ab00, __x=@0x7fa6c235aa50: 0) at /usr/include/c++/7/bits/stl_vector.h:954
#9  0x00007fa6c9511ff1 in Tango::Attribute::fire_change_event (this=0x55ef509fd890, except=0x0) at /home/graziano/Downloads/cppTango-9.3.2/cppapi/server/attribute.cpp:4079
#10 0x00007fa6c953c6ac in Tango::DeviceImpl::push_change_event (this=0x55ef509f4c60, attr_name="doubleattr1", p_data=0x55ef509f6030, x=1, y=0, release=false)
    at /home/graziano/Downloads/cppTango-9.3.2/cppapi/server/dev_event.cpp:843
#11 0x000055ef4face2f2 in TestPushCrash_ns::readthread::run (this=0x55ef509ebd20) at src/readthread.cpp:99
#12 0x00007fa6c8878673 in omni_thread_wrapper () from /usr/local/omniorb-4.2.1/lib/libomnithread.so.4
#13 0x00007fa6c6bd96db in start_thread (arg=0x7fa6c235b700) at pthread_create.c:463
#14 0x00007fa6c800488f in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95

--------------------------------

#0  0x0e46c41c in raise () from /lib/libc.so.6
#1  0x0e46e09c in abort () from /lib/libc.so.6
#2  0x0e4aaea4 in ?? () from /lib/libc.so.6
#3  0x0e4b2ce8 in ?? () from /lib/libc.so.6
#4  0x0e4b5158 in free () from /lib/libc.so.6
#5  0x0e795a0c in operator delete () from /usr/lib/libstdc++.so.6
#6  0x0e795a84 in operator delete[] () from /usr/lib/libstdc++.so.6 
[test_push_crash.zip](https://github.com/tango-controls/cppTango/files/2744418/test_push_crash.zip)

#7  0x0f52a0d4 in _CORBA_Sequence<double>::freebuf (b=0x100b7da0) at /usr/local/omniorb-4.2.1/include/omniORB4/seqTemplatedecls.h:91
#8  0x0f52a150 in ~_CORBA_Sequence (this=0x3dbbbab0) at /usr/local/omniorb-4.2.1/include/omniORB4/seqTemplatedecls.h:129
#9  0x0f52a1b0 in ~_CORBA_Unbounded_Sequence (this=0x3dbbbab0) at /usr/local/omniorb-4.2.1/include/omniORB4/seqTemplatedecls.h:255
#10 0x0f52a204 in ~_CORBA_Unbounded_Sequence_w_FixSizeElement (this=0x3dbbbab0)
    at /usr/local/omniorb-4.2.1/include/omniORB4/seqTemplatedecls.h:518
#11 0x0f52a258 in ~DevVarDoubleArray (this=0x3dbbbab0) at /opt/src/packages/cppTango-9.3.2-nommx/cppapi/server/idl/tango.h:233
#12 0x0f5b5b30 in Tango::AttrValUnion::_release_member (this=0x10140538)
    at /opt/src/packages/cppTango-9.3.2-nommx/cppapi/server/idl/tango.h:3281
#13 0x0f5b6384 in Tango::AttrValUnion::union_no_data (this=0x10140538, _value=true)
    at /opt/src/packages/cppTango-9.3.2-nommx/cppapi/server/idl/tango.h:3204
#14 0x0f5b7028 in Tango::AttrValUnion::operator= (this=0x10140538, _value=@0x1050005c)
    at /opt/src/packages/cppTango-9.3.2-nommx/cppapi/server/idl/tango.h:2878
#15 0x0f8b2c5c in Tango::EventSupplier::detect_and_push_change_event (this=0x100c18e8, device_impl=0x33830008, attr_value=@0x3a8f6a2c,
    attr=@0x10140320, attr_name=@0x3dbbc340, except=0x0, user_push=false)
    at /opt/src/packages/cppTango-9.3.2-nommx/cppapi/server/eventsupplier.cpp:417
#16 0x0f8b3d4c in Tango::EventSupplier::detect_and_push_events (this=0x100c18e8, device_impl=0x33830008, attr_value=@0x3a8f6a2c,
    except=0x0, attr_name=@0x3dbbc340, time_bef_attr=0x3a8f6b4c)
    at /opt/src/packages/cppTango-9.3.2-nommx/cppapi/server/eventsupplier.cpp:179
#17 0x0f90dc84 in Tango::PollThread::poll_attr (this=0x100e8b90, to_do=@0x3a8f6c94)
    at /opt/src/packages/cppTango-9.3.2-nommx/cppapi/server/pollthread.cpp:1970
#18 0x0f910ae4 in Tango::PollThread::one_more_poll (this=0x100e8b90)
    at /opt/src/packages/cppTango-9.3.2-nommx/cppapi/server/pollthread.cpp:800
#19 0x0f9133d8 in Tango::PollThread::run_undetached (this=0x100e8b90, ptr=0x0)
    at /opt/src/packages/cppTango-9.3.2-nommx/cppapi/server/pollthread.cpp:193
#20 0x0e8b6ee8 in omni_thread_wrapper () from /usr/local/omniorb-4.2.1/lib/libomnithread.so.4
#21 0x0ff1fb14 in start_thread () from /lib/libpthread.so.0
#22 0x0e5177d4 in clone () from /lib/libc.so.6

bourtemb, 2019-01-10:
Thank you very much @gscalamera for the detailed bug report and the simple example to reproduce the issue.
I created https://github.com/tango-controls/cppTango/pull/513 Pull Request as an attempt to fix the issue. I am running your TestPushCrash device server with this version of cppTango code since yesterday without any crash.
The code of this PR is the same as the one from github.com/bourtemb/cppTango fix_push_event_in_thread_crash branch I told you about via e-mail earlier today.

liujin001, 2019-01-25:
I ported the code to Windows(edit the readthread.cpp and readthread.h), then compiled with libtango_9.3.3_v140_x64, run server up to 40 minutes on Windows 10, no crash happen.
[test_push_crash_win.zip](https://github.com/tango-controls/cppTango/files/2794785/test_push_crash_win.zip)

gscalamera, 2019-01-28:
Thanks Liu Jin for the feedback.
I slightly modified your code to have some divisions by zero for the attribute value and some ATTR_INVALID for the attribute quality:
[test_push_crash.zip](https://github.com/tango-controls/cppTango/files/2803040/test_push_crash.zip)
You could test again with the code attached.
Note that it has been confirmed to crash in Linux, also with the patch posted by @bourtemb applied, if there are at least one Tango 8 and one Tango 9 events clients receiving events from the test ds.

mliszcz, 2019-11-27:
Hi @gscalamera, @liujin001, @bourtemb,
I started to work on this issue but I need your help with reproduction (on Debian, if possible).
I have following installed:
* cppTango latest tango-9-lts
* DS from Graziano: https://github.com/tango-controls/cppTango/files/2803040/test_push_crash.zip (compiled with `g++ -I../src/ -I/usr/local/include/tango ../src/*.cpp -ltango -lzmq -lomnithread -lomniORB4 -o TestPushCrash`
* tango client (latest tango-9-lts)

I am successfully receiving events on the client side. I've been running the DS without a crash for more than two hours.

Is parallel Tango 8 client really necessary to reproduce the issue? I started another client (with libtango 8 from debian jessie package) and the the callback is only called two times (no err) for each subscription.

Could you please check the client code if it is the same as yours and provide exact tango 8 version needed to reproduce the problem?

Thanks!
Michal

The client:
```cpp
#include <tango.h>
#include <iostream>
#include <unistd.h>
#include <signal.h>

volatile sig_atomic_t stop;

void on_sigint(int signum)
{
    stop = 1;
}

class Callback : public Tango::CallBack
{
    void push_event(Tango::EventData* event)
    {
        /* std::cerr << "got event" << std::endl; */
    }
};

int main(int, char**)
{
    signal(SIGINT, on_sigint);

    try
    {
    Tango::DeviceProxy proxy = Tango::DeviceProxy("test/push_crash/01");
    Callback callback;
    int sub1 = proxy.subscribe_event("doubleattr1", Tango::EventType::CHANGE_EVENT, &callback);
    int sub2 = proxy.subscribe_event("doubleattr2", Tango::EventType::CHANGE_EVENT, &callback);
    int sub3 = proxy.subscribe_event("doubleattr1", Tango::EventType::ARCHIVE_EVENT, &callback);
    int sub4 = proxy.subscribe_event("doubleattr2", Tango::EventType::ARCHIVE_EVENT, &callback);

    while (not stop)
    {
        usleep(10*1000*1000);
    }

    proxy.unsubscribe_event(sub1);
    proxy.unsubscribe_event(sub2);
    proxy.unsubscribe_event(sub3);
    proxy.unsubscribe_event(sub4);

    }
    catch(const Tango::DevFailed& e)
    {
        std::cerr << e.errors[0].desc << std::endl;
    }

    return 0;
}

```

bourtemb, 2019-11-27:
> I am successfully receiving events on the client side. I've been running the DS without a crash for more than two hours.

As far as I remember, it was not so easy to make it crash... I think I increased the polling frequency at some point during my tests but still it was not crashing instantaneously.
 
> Is parallel Tango 8 client really necessary to reproduce the issue? 

As far as I remember, this was necessary indeed to reproduce the issue.
I don't know whether the starting order of the clients is important or not.
I was using tango 8.1.4 when I was working on this issue (on Debian wheezy).

> I started another client (with libtango 8 from debian jessie package) and the the callback is only called two times (no err) for each subscription.

This is an issue indeed. If you wait for 10 seconds, do you get an error event and some re-subscription calls?
If yes, this would indicate that you are facing the famous hearbeat event issue....
If no, it could indicate that the heartbeat events are received but the events themselves are not...
So this could be an issue with the channel name used by ZMQ (mismatch on client and server side).
In Tango 8, this could be solved by using the same TANGO_HOST string on client and server.
Is your clients and server on the same host? Are you using localhost:<port_number> as TANGO_HOST or <ipaddr>:<port>?

Your client code looks good except that the stderr call is commented in the callback, but I guess you know that ;-)...

bourtemb, 2019-11-27:
I managed to reproduce on Debian 9 with https://github.com/tango-controls/cppTango/files/2803040/test_push_crash.zip (modified to have attr_double1 and attr_double2 as member of class readthread to be able to compile) and with the latest version from 9.3-backports branch (latest commit: b187b7e1af70ec0123b8cbc0a506e8c9855c0c80).
It crashed very quickly simply by putting a polling period of 11 ms for doubleattr1 and 9ms for double_attr2 and launching one instance of atkpanel on this device.

Here is the backtrace:
```
[...]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
Core was generated by `bin/debian9/TestPushCrash reynald'.
Program terminated with signal SIGABRT, Aborted.
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:51
51	../sysdeps/unix/sysv/linux/raise.c: No such file or directory.
[Current thread is 1 (Thread 0x7f1d49e14700 (LWP 4553))]
(gdb) bt
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:51
#1  0x00007f1d4e32a42a in __GI_abort () at abort.c:89
#2  0x00007f1d4e366c00 in __libc_message (do_abort=do_abort@entry=2, fmt=fmt@entry=0x7f1d4e45bfd0 "*** Error in `%s': %s: 0x%s ***\n") at ../sysdeps/posix/libc_fatal.c:175
#3  0x00007f1d4e36cfc6 in malloc_printerr (action=3, str=0x7f1d4e45c128 "double free or corruption (fasttop)", ptr=<optimized out>, ar_ptr=<optimized out>) at malloc.c:5049
#4  0x00007f1d4e36d80e in _int_free (av=0x7f1d38000020, p=0x7f1d38002470, have_lock=0) at malloc.c:3905
#5  0x00007f1d5093bbc0 in _CORBA_Sequence<double>::freebuf (b=0x7f1d38002480) at /segfs/tango/ORB/omniORB4.2.3/debian9/include/omniORB4/seqTemplatedecls.h:89
#6  0x00007f1d5093d063 in _CORBA_Sequence<double>::~_CORBA_Sequence (this=0x7f1d38000cc0, __in_chrg=<optimized out>)
    at /segfs/tango/ORB/omniORB4.2.3/debian9/include/omniORB4/seqTemplatedecls.h:127
#7  0x00007f1d5093b4fc in _CORBA_Unbounded_Sequence<double>::~_CORBA_Unbounded_Sequence (this=0x7f1d38000cc0, __in_chrg=<optimized out>)
    at /segfs/tango/ORB/omniORB4.2.3/debian9/include/omniORB4/seqTemplatedecls.h:253
#8  0x00007f1d50939b72 in _CORBA_Unbounded_Sequence_w_FixSizeElement<double, 8, 8>::~_CORBA_Unbounded_Sequence_w_FixSizeElement (this=0x7f1d38000cc0, __in_chrg=<optimized out>)
    at /segfs/tango/ORB/omniORB4.2.3/debian9/include/omniORB4/seqTemplatedecls.h:516
#9  0x00007f1d50936856 in Tango::DevVarDoubleArray::~DevVarDoubleArray (this=0x7f1d38000cc0, __in_chrg=<optimized out>)
    at /users/bourtemb/GIT/tango-controls/cppTango-9.3.3/cppapi/server/idl/tango.h:233
#10 0x00007f1d509969a7 in Tango::AttrValUnion::_release_member (this=0x55ad4446a7c0) at /users/bourtemb/GIT/tango-controls/cppTango-9.3.3/cppapi/server/idl/tango.h:3281
#11 0x00007f1d50996383 in Tango::AttrValUnion::double_att_value (this=0x55ad4446a7c0, _value=...)
    at /users/bourtemb/GIT/tango-controls/cppTango-9.3.3/cppapi/server/idl/tango.h:3036
#12 0x00007f1d50995f23 in Tango::AttrValUnion::operator= (this=0x55ad4446a7c0, _value=...) at /users/bourtemb/GIT/tango-controls/cppTango-9.3.3/cppapi/server/idl/tango.h:2860
#13 0x00007f1d50afc476 in Tango::Attribute::fire_change_event (this=0x55ad4446a310, except=0x0)
    at /users/bourtemb/GIT/tango-controls/cppTango-9.3.3/cppapi/server/attribute.cpp:4046
#14 0x00007f1d50b2450f in Tango::DeviceImpl::push_change_event (this=0x55ad444632a0, attr_name="doubleattr1", p_data=0x55ad44460528, x=1, y=0, release=false)
    at /users/bourtemb/GIT/tango-controls/cppTango-9.3.3/cppapi/server/dev_event.cpp:843
#15 0x000055ad434d7df7 in TestPushCrash_ns::readthread::run (this=0x55ad44460490) at readthread.cpp:131
#16 0x00007f1d4f5a3749 in omni_thread_wrapper () from /segfs/tango/release/debian9/lib/libomnithread.so.4
#17 0x00007f1d4ef394a4 in start_thread (arg=0x7f1d49e14700) at pthread_create.c:456
#18 0x00007f1d4e3ded0f in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:97
```

And here is the backtrace for all the threads:
```
(gdb) thread apply all bt

Thread 12 (Thread 0x7f1d4ae16700 (LWP 4550)):
#0  0x00007f1d4e3df303 in epoll_wait () at ../sysdeps/unix/syscall-template.S:84
#1  0x00007f1d4f366137 in zmq::epoll_t::loop (this=0x55ad44452740) at ../../../src/epoll.cpp:145
#2  0x00007f1d4f3852f0 in thread_routine (arg_=0x55ad444527b0) at ../../../src/thread.cpp:81
#3  0x00007f1d4ef394a4 in start_thread (arg=0x7f1d4ae16700) at pthread_create.c:456
#4  0x00007f1d4e3ded0f in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:97

Thread 11 (Thread 0x7f1d4de59700 (LWP 4538)):
#0  do_sigwait (set=<optimized out>, set@entry=0x7f1d4de58db0, sig=sig@entry=0x7f1d4de58dac) at ../sysdeps/unix/sysv/linux/sigwait.c:64
#1  0x00007f1d4ef42d0d in __sigwait (set=0x7f1d4de58db0, sig=0x7f1d4de58dac) at ../sysdeps/unix/sysv/linux/sigwait.c:96
#2  0x00007f1d50cea388 in Tango::DServerSignal::ThSig::run_undetached (this=0x55ad4443bd70, ptr=0x0) at /users/bourtemb/GIT/tango-controls/cppTango-9.3.3/cppapi/server/thsig.cpp:89
#3  0x00007f1d4f5a3731 in omni_thread_wrapper () from /segfs/tango/release/debian9/lib/libomnithread.so.4
#4  0x00007f1d4ef394a4 in start_thread (arg=0x7f1d4de59700) at pthread_create.c:456
#5  0x00007f1d4e3ded0f in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:97

Thread 10 (Thread 0x7f1d2bfff700 (LWP 5612)):
#0  0x00007f1d4ef4251f in __libc_recv (fd=23, buf=0x7f1d24000d58, n=8192, flags=0) at ../sysdeps/unix/sysv/linux/x86_64/recv.c:28
[... cut ...]
      _pd_state_att_value = 0x7f1d38000b30, _pd_dev_state_att = 939526960, _pd_encoded_att_value = 0x7f1d38000b30, _pd_union_no_data = 48}}, quality = Tango::ATTR_VALID, 
  data_format = Tango::SCALAR, data_type = 5, time = {tv_sec = 1574875050, tv_usec = 179875, tv_nsec = 0}, name = {_ptr = 0x7f1d38001de0 "doubleattr1"}, r_dim = {dim_x = 1, 
    dim_y = 0}, w_dim = {dim_x = 0, dim_y = 0}, err_list = {<_CORBA_Unbounded_Sequence<Tango::DevError>> = {<_CORBA_Sequence<Tango::DevError>> = {pd_max = 0, pd_len = 0, 
        pd_rel = true, pd_bounded = false, pd_buf = 0x0}, <No data fields>}, <No data fields>}}
```

mliszcz, 2019-11-28:
Thanks @bourtemb! With this configuration it crashed for me as well. I will continue the investigation - if you plan to also work on this, please let me know.

mliszcz, 2019-12-03:
Hi all,

I've recompiled tango, omniORB, zmq and TestPushCrash device with thread sanitizer (-fsanitize=thread).

I've then attempted to resolve all possible race conditions in the test scenario (code in #635  , quite dirty, needs refactoring anyway). I've been running the test for two hours without any crash. The test scenario:
* cppTango (from PR #635 ) and TestPushCrash device server (pushing every 0.5ms)
* cppTango from tango-9-lts branch and client posted in previous post
* cppTango from debian 8 package and client posted in previous post
* polling and event thresholds configured to: 11ms/9ms (changed manually as Reynald proposed) and 0.1 (default)

In general, three threads are involved in data races:
* readthread (from TestPushCrash application)
* tango polling thread
* omni orb worker threads (for read_attributes and command_inout)

At least following data races has been addressed in the PR:
* any issues resolved by Reynald in #513
* polling thread (eventsupplier.cpp:1833) and readthread (attribute.cpp:4049) when setting previous attribute value
* polling thread (eventsupplier.cpp:779) and readthread (eventsupplier.cpp:1833) when setting previous attribute value
* readthread (readthread.cpp:116) and polling thread (TestPushCrash.cpp:246) when setting attribute quality and value - **change needed in TestPushCrash**
* omni worker (device_5.cpp:111) and polling thread (same location) while handling read_attributes
* omni worker (eventcmds.cpp :468) and polling thread (pollthread.cpp:1874) when handling event subscription change (get/set use_zmq_event)
* omni worker (eventcmds.cpp:375) and readthread when setting change event timestamps
* readthread (zmqeventsupplier.cpp:1070) and omni worker (zmqeventsupplier:779) when handling event subscription change and setting event_cptr)
* readthread and omni worker when handling event subscription change and setting double_send
* event heartbeat thread (pollthread.cpp:2105) and omni worker (eventcmds.cpp:988) when handling zmq event subscription change amd setting one_subscription_cmd

Attached is tsan log from clean tango-9-lts and with mutex applied to Attribute::fire_change_event / fire_archive_event.

[cppTango-511-mutex-in-attribute-fire-change-event-tsan-stderr.log](https://github.com/tango-controls/cppTango/files/3916498/cppTango-511-mutex-in-attribute-fire-change-event-tsan-stderr.log)
[cppTango-511-no-changes-tsan-stderr.log](https://github.com/tango-controls/cppTango/files/3916499/cppTango-511-no-changes-tsan-stderr.log)

With the changes applied, thread sanitizer is not generating any warnings, except: server startup, server shutdown, some socket access within omniORB.

@gscalamera could you please apply patch from #635 and check if it fixes the crash for you?
Also please put a lock around att_quality and attr_double1 and attr_double2 in your device server, e.g.: (sorry for broken indentation):

readthread.cpp:
```cpp
        {
            omni_mutex_lock oml(access_mutex);
		if(counter % 41 == 0)
			_device->att_quality=Tango::ATTR_INVALID;
		else
			_device->att_quality=Tango::ATTR_VALID;

		attr_double1 = (double)counter/now.tv_usec;
		if(counter % 2 == 0)
		{
			attr_double2 = (double)now.tv_sec/now.tv_usec;
		}
		else
		{
			attr_double2 = (double)now.tv_usec/now.tv_sec;
		}
        }
//...
		        auto quality = Tango::ATTR_VALID;
                {
                    omni_mutex_lock oml(access_mutex);
			        quality = _device->att_quality;
                }
				_device->push_change_event("doubleattr2",&attr_double2, now, quality);
				_device->push_archive_event("doubleattr2",&attr_double2, now, quality);

```

TestPushCrash.cpp:
```cpp
    {
        omni_mutex_lock oml(loop->access_mutex);
	    *attr_doubleattr1_read = loop->attr_double1;
    }
//...
    {
        omni_mutex_lock oml(loop->access_mutex);
	    *attr_doubleattr2_read = loop->attr_double2;
	}
	auto quality = Tango::ATTR_VALID;
    {
        omni_mutex_lock oml(loop->access_mutex);
        quality = att_quality;
    }
	attr.set_value_date_quality(attr_doubleattr2_read, now, quality);

```

gscalamera, 2019-12-03:
Thanks @mliszcz, I will leave the test running for some time.

commit ec898239 (2020-01-24): Merge pull request #641 from mliszcz/fix-511-crash-race-condition-event-push-attempt-2

Correction for race condition between polling threads and user threads pushing events.

Three different data races are addressed in the PR:
* All code that sets previous event value (like in below example) in attribute.cpp and eventsupplier.cpp, is protected by `event_mutex`,
* We remove `detect_mutex`. Previously it was protecting the whole `EventSupplier::detect_change`. Now this method is protected by `event_mutex`,
* Setting timestamps of last event subscription is protected now by `event_mutex`.

Fixes [... cut]
closed 2020-01-24 by mliszcz