tango-controls/cppTango #524: Unstable event::pipe_event testcase (assertion failed)

mliszcz, 2019-01-29:
`event::pipe_event` testcase failed in travis job (Debian 8, Release, zmq 4.2.2):
https://travis-ci.org/tango-controls/cppTango/jobs/485832345

I've restarted the job without touching the code and the issue disappeared.
I haven't investigated this yet. I don't know if this is a Tango bug or some race condition in the testcase.

```
      Start 93: event::pipe_event
93/94 Test #93: event::pipe_event .................***Exception: Other  6.19 sec
   subscribe_event --> OK
   received event --> OK
   received event (with specified date) --> OK
   received event (with error) --> OK
   received event (no copy sending) --> OK
   unsubscribe_event --> OK
pipe_event: /home/tango/src/cpp_test_suite/event/pipe_event.cpp:238: int main(int, char**): Assertion `cb.cb_executed == 2' failed.
```

bourtemb, 2019-02-26:
Hi @mliszcz,

I've spent some time trying to understand the origin of these recurrent test failures in pipe_event test.
I could reproduce the error from time to time.
The tests were failing at 2 different locations:
https://github.com/tango-controls/cppTango/blob/tango-9-lts/cpp_test_suite/event/pipe_event.cpp#L132
and
https://github.com/tango-controls/cppTango/blob/tango-9-lts/cpp_test_suite/event/pipe_event.cpp#L238

What I observed, using Wireshark was that, even though DeviceProxy::subscribe_event() previous call succeeded, the ZMQ subscription message was received later by the publisher, and sometimes. even after the execution of the PushPipeEvent command (case of the [Line132](https://github.com/tango-controls/cppTango/blob/tango-9-lts/cpp_test_suite/event/pipe_event.cpp#L132) failure). So the subscriber could not receive the events because the ZMQ subscription phase was not completed.

This reminded me https://github.com/tango-controls/cppTango/issues/399 and https://github.com/tango-controls/cppTango/pull/415 where we reduced a sleep time during the subscription command.
When we first tested it, the pipe_event test was also failing.
@sergirubio managed to find a sleep time value where the tests would not fail but it looks like the tests have been failing more often recently.
This might be due to the new Travis CI infrastructure.

Here is what the comments above those sleep() calls are saying:
> // Sleep for some mS in order to give to ZMQ some times to propagate the subscription to the publisher

and

> // A small mS sleep here! This is required in case there is a push_event in the read_attribute (or pipe)
// method on the device side. This sleep gives time to ZMQ to send its subscription message

The problem we encounter in pipe_event test is due to that.

I think it would be great to find a way to ensure that when subscribe_event is executed correctly, the ZMQ subscription phase is fully completed on the publisher side when subscribe_event calls returns.

It is not the case today, mainly due to the way ZMQ works...
When the ZMQ call :

`event_sub_sock->setsockopt(ZMQ_SUBSCRIBE,event_name,::strlen(event_name));`

 to tell ZMQ to subscribe to events, succeeds, there is no guarantees the publisher received the subscription message yet. The subscription message is sent asynchronously and it seems like there is no easy way to know when the publisher will receive this message and if it received it.

In ZMQ guide examples, there are even adding a brute-force sleep of 1 second:
http://zguide.zeromq.org/page:all#Node-Coordination

Here is what the zguide says about this example:
>We can't assume that the SUB connect will be finished by the time the REQ/REP dialog is complete. There are no guarantees that outbound connects will finish in any order whatsoever, if you're using any transport except inproc. So, the example does a brute force sleep of one second between subscribing, and sending the REQ/REP synchronization.

>A more robust model could be:

>    Publisher opens PUB socket and starts sending "Hello" messages (not data).
    Subscribers connect SUB socket and when they receive a Hello message they tell the publisher via a REQ/REP socket pair.
    When the publisher has had all the necessary confirmations, it starts to send real data.

This example in the zguide does not fit our use case because in our case, when a subscriber subscribes to events on an attribute, there might already be other subscribers which subscribed for these events so the publisher already started to send real data.

I think it would be interesting to do some tests and see if a handshake mechanism could be implemented during the ZMQ subscription to avoid all these sleep() calls...

t-b, 2019-02-26:
> I think it would be interesting to do some tests and see if a handshake mechanism could be implemented during the ZMQ subscription to avoid all these sleep() calls...

That is definitly the right approach, `sleep(...)` always indicates a hack.

mliszcz, 2019-02-26:
Thanks @bourtemb for detailed analysis!
I've just tried to remove the `sleep()` call from the line above the assertion and I got a reproduction every time I ran the testcase.

Regarding the correction and handshaking at ZMQ/event level, I was rather thinking about using a condition variable to block the testcase until the callback is invoked by ZMQ. Testcase will wait on the condition variable and will proceed with assertions only after explicit notification from callback. Of course some guard timeout is needed when waiting for the callback (5s?).

I have not tried to implement this. I could not figure out any portable (Unix+WIndows) solution without using C++11 threading library.

t-b, 2019-02-26:
> Regarding the correction and handshaking at ZMQ/event level, I was rather thinking about using a condition variable to block the testcase until the callback is invoked by ZMQ. Testcase will wait on [...]

But that would not be a realworld test but some artificial situation. The test case failure points at a potential tango bug.

t-b, 2019-02-26:
The most honest solution, and by far easiest, is to mark the tests with

```
TS_SKIP("Flaky test");
```

and skip them for now.

Ingvord, 2019-02-27:
I agree with @t-b. IMHO nobody really uses pipes due to their tricky API  at least in C++ (I use Pipes extensively in Java though). So `TS_SKIP` is really the honest solution for now.

 In the long term we have to think about handshake as indicated by @bourtemb . But I am worried that it is only feasible to do for new versions i.e. new client <-> new server = OK otherwise unlikely. This is what I tried to do in #385 and #404

bourtemb, 2019-02-27:
> Regarding the correction and handshaking at ZMQ/event level, I was rather thinking about using a condition variable to block the testcase until the callback is invoked by ZMQ.
> Testcase will wait on [...]

This is more tricky than it seems.
The callback is actually always invoked by ZMQ during the subscribe_event call (Tango will call the callback with the result of the synchronous read call). But the first time the callback is called, there are no guarantees that the publisher already received the ZMQ subscription call. And we cannot wait for the callback to be called a second time because in our testcase, when the pipe event is pushed by code from the read pipe method, the ZMQ publisher might not have received the ZMQ subscription call yet, so in this case, the pipe event pushed by code might not be sent at all to the new subscriber, so the assertions might still fail.

> But that would not be a realworld test but some artificial situation. The test case failure points at a potential tango bug.

Indeed. I agree with @t-b. 
There is no guarantee that after a successful call to subscribe_event() that a subscriber will receive all the events it is interested in. 
In some situations, some important events might be missed if they occur between the subscribe_event() call and the ZMQ subscription reception on the ZMQ publisher side, potentially leading to situations where a client application might show out of date/incorrect values.

This problem can affect the pipe_event test but I think it can potentially affect other tests. I saw some other random test failures in some other event related tests where sometimes the subscribers do not receive the expected number of events.

Indeed, @ingvord, the handshake mechanism can be implemented only for the case new client <=> new server.

Ingvord, 2019-08-21:
@mliszcz , @t-b , @bourtemb 

I have played a little bit with the code in #589 

And here are some of my thoughts:

1) If we want to get rid off sleeps that basically means in the code below `device->subscribe_event` becomes a blocking function which is blocked till the client executes first callback. This is technically achievable but will rely on sync call. Which I am strictly against (there must not be any sync calls during subscription what so ever, but that is a different story)

```cpp
//
// subscribe to a pipe event
//
		int eve_id1 = device->subscribe_event("RWPipe",Tango::PIPE_EVENT,&cb);
//
// The callback should have been executed once
//
		assert (cb.cb_executed == 1);

```

2) due to 1) I would use conditional variables in callback functions to fix this (and maybe others) test(s). I think it is OK, even though C++11 etc as this change will affect only tests and not library code

3) We can think about improving hand-shake, but I do not see clear benefit vs complexity. The nature of PUB/SUB is that SUB lives its own live sending events, SUBs occasionally appear/disappear receiving some of the events. It MUST be OK for a SUB to miss some events 

I will continue with conditional variables approach, while the rest (especially 3) requires severe discussion

mliszcz, 2019-08-29:
Hi @Ingvord,
sorry for the late comment but last week I was on vacation with no access to my pc.

Have you checked the #558 PR I submitted some time ago? It is similar to what you propose but more generic (and does not expose synchronization code directly to the testcase).

So your example code becomes:

```cpp
//
// subscribe to a pipe event
//
		int eve_id1 = device->subscribe_event("RWPipe",Tango::PIPE_EVENT,&cb);
//
// The callback should have been executed once
//
		// assert (cb.cb_executed == 1);
		int timeout = 500;
		cb.assert_called(timeout); // waits up to 500ms for the call 
```

mliszcz, 2019-08-29:
One more comment:
> 3. We can think about improving hand-shake, but I do not see clear benefit vs complexity. The nature of PUB/SUB is that PUB lives its own live sending events, SUBs occasionally appear/disappear receiving some of the events. It MUST be OK for a SUB to miss some events

If I understand correctly, the sole purpose of the hand-shake enhancement is to ensure that subscribe_event returns after callback has been invoked for the first time? If yes, this probably can be resolved by simply changing just the SUB side - intercept the callback and notify some conditional variable.

closed 2022-02-24 by t-b