tango-controls/cppTango #532: Fix doxygen warnings

t-b, 2019-02-07:
```
warning: doxygen no longer ships with the FreeSans font.
You may want to clear or change DOT_FONTNAME.
Otherwise you run the risk that the wrong font is being used for dot generated graphs.
warning: tag INPUT: input source `/home/firma/devel/cppTango/cppapi/server/tango_const.h' does not exist
warning: source /home/firma/devel/cppTango/cppapi/server/tango_const.h is not a readable file or directory... skipping.
/home/firma/devel/cppTango/cppapi/client/group.h:849: warning: found */ without matching /* while inside a \code block! Perhaps a missing \endcode?

Searching for include files...
Searching for example files...
Searching for images...
Searching for dot files...
Searching for msc files...
Searching for dia files...
Searching for files to exclude
Searching INPUT for files to process...
Searching for files in directory /home/firma/devel/cppTango/log4tango
Reading and parsing tag files
Parsing files
Preprocessing /home/firma/devel/cppTango/cppapi/server/device.h...
Parsing file /home/firma/devel/cppTango/cppapi/server/device.h...
Preprocessing /home/firma/devel/cppTango/cppapi/server/deviceclass.h...
Parsing file /home/firma/devel/cppTango/cppapi/server/deviceclass.h...
Preprocessing /home/firma/devel/cppTango/cppapi/server/command.h...
Parsing file /home/firma/devel/cppTango/cppapi/server/command.h...
Preprocessing /home/firma/devel/cppTango/cppapi/server/utils.h...
Parsing file /home/firma/devel/cppTango/cppapi/server/utils.h...
Preprocessing /home/firma/devel/cppTango/cppapi/server/except.h...
Parsing file /home/firma/devel/cppTango/cppapi/server/except.h...
Preprocessing /home/firma/devel/cppTango/cppapi/server/attrdesc.h...
Parsing file /home/firma/devel/cppTango/cppapi/server/attrdesc.h...
Preprocessing /home/firma/devel/cppTango/cppapi/server/fwdattrdesc.h...
Parsing file /home/firma/devel/cppTango/cppapi/server/fwdattrdesc.h...
Preprocessing /home/firma/devel/cppTango/cppapi/server/pipedesc.h...
Parsing file /home/firma/devel/cppTango/cppapi/server/pipedesc.h...
Preprocessing /home/firma/devel/cppTango/cppapi/server/attribute.h...
Parsing file /home/firma/devel/cppTango/cppapi/server/attribute.h...
Preprocessing /home/firma/devel/cppTango/cppapi/server/multiattribute.h...
Parsing file /home/firma/devel/cppTango/cppapi/server/multiattribute.h...
Preprocessing /home/firma/devel/cppTango/cppapi/server/pollext.h...
Parsing file /home/firma/devel/cppTango/cppapi/server/pollext.h...
Preprocessing /home/firma/devel/cppTango/cppapi/server/device_2.h...
Parsing file /home/firma/devel/cppTango/cppapi/server/device_2.h...
Preprocessing /home/firma/devel/cppTango/cppapi/server/w_attribute.h...
Parsing file /home/firma/devel/cppTango/cppapi/server/w_attribute.h...
Preprocessing /home/firma/devel/cppTango/cppapi/server/pipe.h...
Parsing file /home/firma/devel/cppTango/cppapi/server/pipe.h...
Preprocessing /home/firma/devel/cppTango/cppapi/server/w_pipe.h...
Parsing file /home/firma/devel/cppTango/cppapi/server/w_pipe.h...
Preprocessing /home/firma/devel/cppTango/cppapi/server/device_3.h...
Parsing file /home/firma/devel/cppTango/cppapi/server/device_3.h...
Preprocessing /home/firma/devel/cppTango/cppapi/server/device_4.h...
Parsing file /home/firma/devel/cppTango/cppapi/server/device_4.h...
Preprocessing /home/firma/devel/cppTango/cppapi/server/device_5.h...
Parsing file /home/firma/devel/cppTango/cppapi/server/device_5.h...
Preprocessing /home/firma/devel/cppTango/cppapi/server/encoded_attribute.h...
Parsing file /home/firma/devel/cppTango/cppapi/server/encoded_attribute.h...
Preprocessing /home/firma/devel/cppTango/cppapi/server/attrprop.h...
Parsing file /home/firma/devel/cppTango/cppapi/server/attrprop.h...
Preprocessing /home/firma/devel/cppTango/cppapi/client/dbapi.h...
Parsing file /home/firma/devel/cppTango/cppapi/client/dbapi.h...
Preprocessing /home/firma/devel/cppTango/cppapi/client/apiexcept.h...
Parsing file /home/firma/devel/cppTango/cppapi/client/apiexcept.h...
Preprocessing /home/firma/devel/cppTango/cppapi/client/devapi.h...
Parsing file /home/firma/devel/cppTango/cppapi/client/devapi.h...
Preprocessing /home/firma/devel/cppTango/cppapi/client/doc.h...
Parsing file /home/firma/devel/cppTango/cppapi/client/doc.h...
Preprocessing /home/firma/devel/cppTango/cppapi/client/devasyn.h...
Parsing file /home/firma/devel/cppTango/cppapi/client/devasyn.h...
Preprocessing /home/firma/devel/cppTango/cppapi/client/event.h...
[... 219 lines cut ...]
Generating docs for compound Tango::UserDefaultPipeProp...
Generating docs for compound Tango::Util...
Generating docs for compound Tango::WAttribute...
Generating docs for compound Tango::WPipe...
Generating graph info page...
Generating directory documentation...
Generating index page...
Generating page index...
Generating module index...
Generating namespace member index...
Generating annotated compound index...
Generating alphabetical compound index...
Generating hierarchical class index...
Generating graphical class hierarchy...
Generating member index...
Generating file member index...
Generating example index...
finalizing index lists...
writing tag file...
Running plantuml with JAVA...
Running dot...
lookup cache used 4568/65536 hits=49300 misses=4807
finished...
Built target doc
```

bourtemb, 2019-03-25:
Almost all the Doxygen warnings should be fixed in PR #544 
@t-b, could you please review this PR?
Only the following warning should remain:
`/home/firma/devel/cppTango/cppapi/client/group.h:849: warning: found */ without matching /* while inside a \code block! Perhaps a missing \endcode?`

It is due to the following code example in the Doxygen documentation:
```
 * @code
 * root_group->remove("*/stepper_motor/*");
 * @endcode
```

Doxygen is not clever enough on this one.

bourtemb, 2019-03-29:
Hi @t-b,
Can we consider this issue solved?

As I wrote before:

> Only the following warning should remain:
> /home/firma/devel/cppTango/cppapi/client/group.h:849: warning: found */ without matching /* while inside a \code block! Perhaps a missing \endcode?
>
> It is [...]

I commented the doxygen ticket you mentioned (https://github.com/doxygen/doxygen/issues/6737) giving the above example.
I think the solution to remove this warning should come from the Doxygen team.

closed 2019-03-29 by t-b