tango-controls/cppTango #534: Cleanup and unify branches

t-b, 2019-02-09:
- [ ] The `tango-9-lts` branches are not required anymore as we don't use master at all.
- [ ] Replace master branch contents with `tango-9-lts` without rewriting it just using `git revert`

- Agree on a common branch  naming scheme for all tango related repos.

Ingvord, 2019-02-10:
Concerning the second point - rewriting will make history much cleaner and current master's history has no value. So I would just reset master to the point when `tango-9-lts` was created and then rebase `tango-9-lts` onto it.

I am not sure about the third one - I mean for me we do not need it. Branches are barely visible in GitHub. My point is that any reasonable name when branch is created will do the job. And most of the repo are just single maintainer repo most of the time even without any external PRs. So it looks like "here are the rules you have to follow to create branches" - "well, that is fine but I do not have branches, and even if I have nobody see them", wierd....

The root cause of this discussion is that:

1) at some point we have decided to have `tango-9-lts` for stable version and `master` for development. But then we realized that 1 - we have no resources to support both; 2 - nobody interested in having branch with current development

2) we did not delete branches after merging PRs that's why we have them so many. So we just need to delete them now and in the future

bourtemb, 2019-02-18:
It's true the _master_ branch is currently not used anymore and this can be confusing for newcomers.
The current _master_ branch corresponds to the Tango V10 development work which was done in the past.
This development has been temporarily stopped and will restart at some point.
I think the work done on Tango V10 in the master branch should be moved to another branch so we could eventually refer to it at some point in the future.
tango-9-lts development could eventually be moved to master while we wait for Tango V10 development to be restarted... But then _tango-9-lts_ branch will have to be created again at some point because it is an LTS version, so by definition it requires its own branch.
So I am not really convinced this is a good idea to move _tango-9-lts_ to _master_.
_tango-9-lts_ branch is currently the default git branch so, when a user goes to github repository, this is the branch which is displayed.
When a user clones cppTango repository, it is in _tango-9-lts_ branch by default, not master.
When there is a Pull Request, Github proposes to merge to _tango-9-lts_ by default too.

So I would prefer to stay as it is right now and to use the _master_ branch for Tango V10 development, and to keep _tango-9-lts_ as default branch until V10 becomes stable...

I am in favour of deleting the branches corresponding to the already merged PRs.

t-b, 2019-06-05:
Closing for now as we want to keep the current branch layout.

closed 2019-06-05 by t-b